nightscout/AndroidAPS #1757: NSCLIENT - Pause

Atlas-Night-Out, 2022-05-21:
Reporting bugs
--------------
NSCLIENT Pauseed keeps coming on, on its own! I have to manualy take it off to keep things running.
[image]

Started 3 weeks ago roughly beginnign of May 20222 
I had not done anything to make this bug happen it started on its own!
Yesterday it got even worse and no matter how many times I too the paused back off it kept coming back on and effecting my readings going to Xdrip and Nightscout 
I have asked in teh group and no one seems to have a fix to this
Im running AndroidAPS 3.0.0.1 and Nightscout Version 14.2.5
[AndroidAPS[1].log](https://github.com/nightscout/AndroidAPS/files/8747381/AndroidAPS.1.log)

MilosKozak, 2022-05-21:
NSClient is puased automaticaly after 5 reconnections within 2 min to prevent high data usage.
it's unpaused after 15 min (but can stay paused if the app is restarted meanwhile)
you have to resolve connectivity issues
btw reconnection is done with cellular <-> wifi change too

closed 2022-05-21 by MilosKozak

Atlas-Night-Out, 2022-05-21:
I have wifi connected all of the time and the pause does not come back on as I see it auto pause and Im connected to wifi!

Atlas-Night-Out, 2022-05-21:
Thank you was the cellular connections nice to know it works this way!