nightscout/AndroidAPS #1758: Profile switch: Percentage missing in Nightscout entry

philipgo, 2022-05-21:
I am using AAPS dev (currently build 36a4ca33e-2022.05.20-18:11). Since around 12.04.2022 profile switches are displayed on Nightscout without the percentage. 

There are no relevant recent changes in Nightscout, so I assume it must have something to do with AndroidAPS.

[image]

The MongoDB entries show that the percentage is missing in the treatment entry.

[image]

MilosKozak, 2022-05-25:
fixed by https://github.com/nightscout/AndroidAPS/commit/4debc1b1708686d38e8544ac82da49b5e2bbbdf1

closed 2022-05-25 by MilosKozak

philipgo, 2022-05-25:
Thank you, Milos!