nightscout/AndroidAPS #1760: Last Dev crash at start

macarolobe, 2022-05-22:
Last dev crash systematically at start. I tried clean install : desintall aaps, delete folder aaps, clean cache.
I'm with Samsung s10e android 12 One UI 4.1 with omnipod dash.
I think the crash appears with the last update for dash.
[AndroidAPS._2022-05-22_17-55-49_.0.zip](https://github.com/nightscout/AndroidAPS/files/8749765/AndroidAPS._2022-05-22_17-55-49_.0.zip)

MilosKozak, 2022-05-22:
there was a recent change which had to be done on pod change. isn't it this case?

MilosKozak, 2022-05-22:
```
17:53:36.500 [main] E/PUMPCOMM: [OmnipodDashPodStateManagerImpl.load():699]: Failed to deserialize Pod state
com.google.gson.JsonSyntaxException: java.lang.NumberFormatException: For input string: "01G3P01WF8F5SZ1GJP128E1MHN"
	at com.google.gson.internal.bind.TypeAdapters$11.read(TypeAdapters.java:320) ~[na:0.0]
	at com.google.gson.internal.bind.TypeAdapters$11.read(TypeAdapters.java:310) ~[na:0.0]
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$1.read(ReflectiveTypeAdapterFactory.java:130) ~[na:0.0]
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:221) ~[na:0.0]
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$1.read(ReflectiveTypeAdapterFactory.java:130) ~[na:0.0]
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:221) ~[na:0.0]
	at com.google.gson.Gson.fromJson(Gson.java:991) ~[na:0.0]
	at com.google.gson.Gson.fromJson(Gson.java:956) ~[na:0.0]
	at com.google.gson.Gson.fromJson(Gson.java:905) ~[na:0.0]
	at com.google.gson.Gson.fromJson(Gson.java:876) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.pod.state.OmnipodDashPodStateManagerImpl.load(OmnipodDashPodStateManagerImpl.kt:694) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.pod.state.OmnipodDashPodStateManagerImpl.<init>(OmnipodDashPodStateManagerImpl.kt:44) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.pod.state.OmnipodDashPodStateManagerImpl_Factory.newInstance(OmnipodDashPodStateManagerImpl_Factory.java:50) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.pod.state.OmnipodDashPodStateManagerImpl_Factory.get(OmnipodDashPodStateManagerImpl_Factory.java:40) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.pod.state.OmnipodDashPodStateManagerImpl_Factory.get(OmnipodDashPodStateManagerImpl_Factory.java:13) ~[na:0.0]
	at dagger.internal.DoubleCheck.get(DoubleCheck.java:47) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.OmnipodDashManagerImpl_Factory.get(OmnipodDashManagerImpl_Factory.java:46) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.OmnipodDashManagerImpl_Factory.get(OmnipodDashManagerImpl_Factory.java:14) ~[na:0.0]
	at dagger.internal.DoubleCheck.get(DoubleCheck.java:47) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.OmnipodDashPumpPlugin_Factory.get(OmnipodDashPumpPlugin_Factory.java:94) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.OmnipodDashPumpPlugin_Factory.get(OmnipodDashPumpPlugin_Factory.java:25) ~[na:0.0]
	at dagger.internal.DoubleCheck.get(DoubleCheck.java:47) ~[na:0.0]
	at dagger.internal.MapFactory.get(MapFactory.java:58) ~[na:0.0]
	at dagger.internal.MapFactory.get(MapFactory.java:31) ~[na:0.0]
	at dagger.internal.DoubleCheck.get(DoubleCheck.java:47) ~[na:0.0]
	at info.nightscout.androidaps.di.AppModule.providesPlugins(AppModule.kt:57) ~[na:0.0]
	at info.nightscout.androidaps.di.AppModule_ProvidesPluginsFactory.providesPlugins(AppModule_ProvidesPluginsFactory.java:75) ~[na:0.0]
	at info.nightscout.androidaps.di.DaggerAppComponent$AppComponentImpl.listOfPluginBase(DaggerAppComponent.java:27494) ~[na:0.0]
	at info.nightscout.androidaps.di.DaggerAppComponent$AppComponentImpl.injectMainApp(DaggerAppComponent.java:31104) ~[na:0.0]
	at info.nightscout.androidaps.di.DaggerAppComponent$AppComponentImpl.inject(DaggerAppComponent.java:31090) ~[na:0.0]
	at info.nightscout.androidaps.di.DaggerAppComponent$AppComponentImpl.inject(DaggerAppComponent.java:26058) ~[na:0.0]
	at dagger.android.DaggerApplication.injectIfNecessary(DaggerApplication.java:63) ~[na:0.0]
	at dagger.android.DaggerApplication.onCreate(DaggerApplication.java:38) ~[na:0.0]
	at info.nightscout.androidaps.MainApp.onCreate(MainApp.kt:92) ~[na:0.0]
	at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1211) ~[na:0.0]
	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:7506) ~[na:0.0]
	at android.app.ActivityThread.access$1700(ActivityThread.java:310) ~[na:0.0]
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2281) ~[na:0.0]
	at android.os.Handler.dispatchMessage(Handler.java:106) ~[na:0.0]
	at android.os.Looper.loopOnce(Looper.java:226) ~[na:0.0]
	at android.os.Looper.loop(Looper.java:313) ~[na:0.0]
	at android.app.ActivityThread.main(ActivityThread.java:8663) ~[na:0.0]
	at java.lang.reflect.Method.invoke(Native Method) ~[na:0.0]
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:567) ~[na:0.0]
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135) ~[na:0.0]
Caused by: java.lang.NumberFormatException: For input string: "01G3P01WF8F5SZ1GJP128E1MHN"
	at sun.misc.FloatingDecimal.readJavaFormatString(FloatingDecimal.java:2043) ~[na:0.0]
	at sun.misc.FloatingDecimal.parseDouble(FloatingDecimal.java:110) ~[na:0.0]
	at java.lang.Double.parseDouble(Double.java:538) ~[na:0.0]
	at com.google.gson.stream.JsonReader.nextLong(JsonReader.java:964) ~[na:0.0]
	at com.google.gson.internal.bind.TypeAdapters$11.read(TypeAdapters.java:318) ~[na:0.0]
	... 44 common frames omitted
```

MilosKozak, 2022-05-22:
i'd suggest load previous version and upgrade on pod change

macarolobe, 2022-05-22:
I have to deactivate pod before the update ?

MilosKozak, 2022-05-22:
yes

macarolobe, 2022-05-23:
I disabled the pod before the update but same effect, the app crash at the start. I will retry next pod change.

macarolobe, 2022-05-26:
So after a lot of tries with crash at start, it seems that the Wear module is the problem. When I disabled it the installation was ok with no crash. So I will try at my next pod change.

mick711, 2022-05-26:
Thanks i juste today AAPS doesn't start i need to force for Time to start , i don't know why but i use Wear. @`
[Uploading AndroidAPS.log…]()
`

MilosKozak, 2022-05-26:
i see crashes related to not updated wear app

macarolobe, 2022-05-30:
After disabled wear module, no problem to install dev and new pod.

mick711, 2022-05-31:
I think this issue can be closed . All work fine after update wear, now i know when i update i need to update app and wear .

avereha, 2022-06-17:
I don't think this issue is dash-specific. 
The error in the logs in inside a try-catch
```
try {
                return Gson().fromJson(
                    sharedPreferences.getString(R.string.key_omnipod_dash_pod_state, ""),
                    PodState::class.java
                )
            } catch (ex: Exception) {
                logger.error(LTag.PUMPCOMM, "Failed to deserialize Pod state", ex)
            }
```
It should not cause a crash

macarolobe, 2022-06-17:
Crashs were cause by the wear module not by dash.

closed 2022-06-19 by MilosKozak