nightscout/AndroidAPS #1764: [dev-k] Autotune throws FATAL EXCEPTION

swissalpine, 2022-05-23:
@Philoul Is this interesting or helpful?
```
2022-05-23 09:48:31.304 2930-2930/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: info.nightscout.androidaps, PID: 2930
    java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.content.Context.isUiContext()' on a null object reference
        at android.view.ViewConfiguration.get(ViewConfiguration.java:502)
        at android.view.View.<init>(View.java:5335)
        at android.view.View.<init>(View.java:5485)
        at android.view.ViewGroup.<init>(ViewGroup.java:699)
        at android.widget.LinearLayout.<init>(LinearLayout.java:254)
        at android.widget.LinearLayout.<init>(LinearLayout.java:250)
        at android.widget.LinearLayout.<init>(LinearLayout.java:246)
        at android.widget.LinearLayout.<init>(LinearLayout.java:242)
        at android.widget.TableLayout.<init>(TableLayout.java:91)
        at info.nightscout.androidaps.plugins.general.autotune.AutotuneFragment.showResults$lambda-41$lambda-40(AutotuneFragment.kt:391)
        at info.nightscout.androidaps.plugins.general.autotune.AutotuneFragment.$r8$lambda$ESyvUDAdq5HElXihWjE_BWA88Bs(Unknown Source:0)
        at info.nightscout.androidaps.plugins.general.autotune.AutotuneFragment$$ExternalSyntheticLambda2.run(Unknown Source:2)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:223)
        at android.app.ActivityThread.main(ActivityThread.java:7700)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:612)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:997)
```

Philoul, 2022-05-23:
Of course I'm interrested, thank's... Are you on latest update of dev branch, and can you reproduce this Fatal Exception ? (on first analysis I don't see what could be wrong, and I never got this crash...)...
@MilosKozak can `context` be null here ? other ideas ?
```
binding.autotuneResults.addView(
                        TableLayout(context).also { layout ->
```

swissalpine, 2022-05-23:
Yes, latest dev, updated this morning.
I think the problem occurs when I go back to the main screen. Before one of your recent fixes I had occasional crashes when doing this, today I had logcat active by chance when the error occurred. However, I have tried a few times now to provoke the error - but this time no exception.

Philoul, 2022-05-23:
And do you have autotune as a tab or in hambuger menu?

swissalpine, 2022-05-23:
Hamburger Menu

Philoul, 2022-05-23:
Ok, thanks.
I'll switch from tab to humburger menu to see if I can reproduce this crash...

swissalpine, 2022-05-23:
I had a crash again. But without Logcat. But the last AAPS log entries:
```
|	23	| 0.850 |0.994	| 17% / 0
+------------------------------------------
|	∑	|	17.5 |	17.9 |
+------------------------------------------

15:06:10.398 [Thread-1968] D/info.nightscout.androidaps.plugins.general.autotune.AutotunePlugin: [FS] Create autotune.2022-05-23.log file in autotune folder
15:06:10.404 [Thread-1968] D/info.nightscout.androidaps.plugins.general.autotune.AutotunePlugin: [FS] Create autotune.2022-05-23.log file in autotune folder
15:06:10.417 [Thread-1968] D/info.nightscout.androidaps.plugins.general.autotune.AutotunePlugin: [FS] Create autotune_recommendations.log file in autotune folder
15:06:10.459 [Thread-1968] D/info.nightscout.androidaps.plugins.general.autotune.AutotunePlugin: [FS] Create autotune_2022-05-23_15-06-08.zip file in /storage/emulated/0/AAPS/logs/info.nightscout.androidaps folder
15:06:10.870 [RxCachedThreadScheduler-47] D/NSCLIENT: [NSClientPlugin.onStart$lambda-5():124]: PING received
15:06:10.871 [RxCachedThreadScheduler-47] D/NSCLIENT: [NSClientPlugin.onStart$lambda-5():124]: QUEUE Resend started: Ping received
15:06:10.894 [RxCachedThreadScheduler-47] D/NSCLIENT: [NSClientPlugin.onStart$lambda-5():124]: QUEUE Resend ended: Ping received
15:06:27.633 [main] D/CONSTRAINTS: [Constraint.setIfSmaller():47]: Setting because of smaller value 1111111.0 -> 12.0 (Limiting IOB to 12.0 U because of max value in preferences)[Safety]
15:06:29.136 [main] D/CORE: [MainApp.onCreate():93]: onCreate
```
Again it happened while going back to the main screen.

MilosKozak, 2022-05-25:
sometimes this happen when UI update is triggered from RX. Event is received after destroying fragment/activity/dialog.
you can check context/_binding at start of function

closed 2022-05-26 by MilosKozak