nightscout/AndroidAPS #1766: [NSClient 3.0.0.2-Master] Problem using NSClient 3.0.0.2 from the latest Master version

naiaguillo, 2022-05-23:
Bugs

NSClient: 3.0.0.2
Nightscout: 14.2.2 and 14.2.5
APPS 3.0.0.2-Master

Everything works fine when using AAPS version 3.0.0.2 linked to Nigtscout (versions 14.2.2 or 14.2.5). The problem occurs when NSClient version 3.0.0.2 is installed on another phone and the master profile is imported. When it starts to sync the data, the Nightscout page crashes and needs to be restarted to get it working again and pause the NSClient sync in the NSClient 3.0.0.2 app.

These are the errors that appear in the log of the Nightscout page on Heroku:

May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620267&selected=1465446609725620267) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620267&selected=1465446609725620267)     bolusCalculatorResult: '{"basalIOB":0.385,"bolusIOB":-0.084,"carbs":11.0,"carbsInsulin":0.7857142857142857,"cob":0.0,"cobInsulin":0.0,"dateCreated":1653326849954,"glucoseDifference":-21.0,"glucoseInsulin":-0.24705882352941178,"glucoseTrend":0.0,"glucoseValue":77.0,"ic":14.0,"id":2,"interfaceIDs_backing":{"nightscoutId":"6288c85debc11c000449eba8"},"isValid":true,"isf":85.0,"note":"","otherCorrection":0.0,"percentageCorrection":100,"profileName":"00 Normoglucemia ","superbolusInsulin":0.0,"targetBGHigh":98.0,"targetBGLow":98.0,"timestamp":1653131334013,"totalInsulin":0.84,"trendInsulin":0.0,"utcOffset":7200000,"version":0,"wasBasalIOBUsed":true,"wasBolusIOBUsed":true,"wasCOBUsed":false,"wasGlucoseUsed":true,"wasSuperbolusUsed":false,"wasTempTargetUsed":false,"wasTrendUsed":false,"wereCarbsUsed":false}',
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620268&selected=1465446609725620268) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620268&selected=1465446609725620268)     date: 1653131334013,
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620269&selected=1465446609725620269) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620269&selected=1465446609725620269)     glucose: 77,
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620270&selected=1465446609725620270) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620270&selected=1465446609725620270)     units: 'mg/dl',
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620271&selected=1465446609725620271) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620271&selected=1465446609725620271)     notes: ''
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620272&selected=1465446609725620272) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620272&selected=1465446609725620272)   }
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620273&selected=1465446609725620273) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620273&selected=1465446609725620273) }
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620274&selected=1465446609725620274) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620274&selected=1465446609725620274) got data-received event, requesting reload
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620275&selected=1465446609725620275) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620275&selected=1465446609725620275) WS: dbUpdate client ID:  qvBHRj7sJ-RrZPokAAAC  data:  {
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620276&selected=1465446609725620276) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620276&selected=1465446609725620276)   collection: 'treatments',
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620277&selected=1465446609725620277) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620277&selected=1465446609725620277)   _id: '628883f82819bc00042463fe',
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620278&selected=1465446609725620278) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620278&selected=1465446609725620278)   data: {
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620279&selected=1465446609725620279) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620279&selected=1465446609725620279)     created_at: '2022-05-21T06:17:06.051Z',
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620280&selected=1465446609725620280) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620280&selected=1465446609725620280)     enteredBy: 'openaps://AndroidAPS',
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620281&selected=1465446609725620281) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620281&selected=1465446609725620281)     eventType: 'Temp Basal',
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620282&selected=1465446609725620282) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620282&selected=1465446609725620282)     isValid: true,
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620283&selected=1465446609725620283) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620283&selected=1465446609725620283)     duration: 25,
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620284&selected=1465446609725620284) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620284&selected=1465446609725620284)     durationInMilliseconds: 1503134,
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620285&selected=1465446609725620285) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620285&selected=1465446609725620285)     type: 'NORMAL',
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620286&selected=1465446609725620286) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620286&selected=1465446609725620286)     rate: 0.2,
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620287&selected=1465446609725620287) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620287&selected=1465446609725620287)     percent: -50,
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620288&selected=1465446609725620288) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620288&selected=1465446609725620288)     pumpId: 394606,
May 23 10:28:40 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446609725620289&selected=1465446609725620289) [app/web.1](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aapp%2Fweb.1&focus=1465446609725620289&selected=1465446609725620289)     endId: 394608,
[... cut ...] selected=1465446683773886465)" dyno= connect= service= status=503 bytes= protocol=https
May 23 10:28:58 [naiaguillo](https://my.papertrailapp.com/systems/naiaguillo/events?focus=1465446685154021498&selected=1465446685154021498) [heroku/router](https://my.papertrailapp.com/groups/33577731/events?q=program%3Aheroku%2Frouter&focus=1465446685154021498&selected=1465446685154021498) at=error code=H10 desc="App crashed" method=GET path="/socket.io/?EIO=3&transport=polling" host=naiaguillo.herokuapp.com

MilosKozak, 2022-05-26:
what do you mean by master profile is imported?
i tested it now on fresh ns and fresh nsclient without issue

fjpezuela, 2022-05-27:
Yes, with the current Master profile imported into an NSClient installation built with the same Keystore as the Master version.

MilosKozak, 2022-05-27:
do not import master data into nsclient
just configure it for NS address

fjpezuela, 2022-05-31:
I have tried without importing the profile and after putting the data of his Nightscout page, a few minutes later, NSClient returns an error and the Nightscout page crashes.

[image]

closed 2022-06-08 by MilosKozak