nightscout/AndroidAPS #1768: Wrong IOB, because boluses are shown twice

Milahan, 2022-05-23:
In Androidaps 3.0, some meal-boluses and SMB's are shown twice in the treatments-tab. The pump (Medtronic 754) shows the boluses only once.
Because of this, the IOB in Androidaps is wrong (too high).

I haven't connected AAPS to NS.
In previous AAPS-versions (<3.0), I've never seen this behaviour.

Here are two examples:

[image]
[image]

ArthurusDent, 2022-05-23:
Please add your setup (smartphone, Android version, exact AndroidAPS version (upper right corner, three dot menu, About, build number) and logs.

andyrozman, 2022-05-23:
Haven't seen double bolus problem since version 2.3 (I think), where it was fixed... You are the first person reporting this in new version... Very weird...

old-square-eyes, 2022-05-26:
Same issue I think.

Slight panic thinking I got double dosed. But pump confirms not. See the two 1.9U Wizard boluses in the treatment log (also double 0.4 SMB), and double IOB. 522 OL.

[image]
ed
[image]

[image]
.

Milahan, 2022-06-10:
This morning, I saw there was another 'double bolus-situation' last night. This time, the boluses were not exactly at the same time (a few minutes apart), according to AAPS. The pump-history shows the bolus only once.

I attached the log, screenshot and photo. I hope this helps in finding the cause.

[image]

[image]

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8876160/AndroidAPS.log)

Milahan, 2022-06-11:
Here's another example from last night (10-06-2022, 19:55-20:00). The pump-history shows the bolus only once.
I also included the complete log file. 
The previous log file might have been a damaged file (?).

[image]

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8883324/AndroidAPS.log)

andyrozman, 2022-06-11:
Is there a problem with your communication? It seems that Bolus was placed 4 minutes before pump started it... 

We pair boluses from pump history with local bolus, and difference can be only 1:59 minutes. 

It seems like your communication is very spotty... Can you confirm that?

Milahan, 2022-06-11:
Thanks for your quick response, Andy!
 
I'm not experiencing connection problems. 
The Rileylink is always near my phone and pump. 
It sometimes does take about 20 seconds for AAPS to make the pump start a bolus. I've never noticed it taking >2 minutes. But if the log says it took so long, it probably did..

If thats the problem: would it be possible to automaticaly delete all past bolusses in AAPS, if they are not present in the pump-history?

andyrozman, 2022-06-11:
Deleting anything automatically could be very dangerous. Question is why it took so long for bolus to come to pump. If you are using OL, using firmware 2.5 , might help you... 

They have rewritten their firmware with version 3.x and while we support it, I think that 2.x communication was more stable.

Milahan, 2022-06-11:
I'm using the (older) Rileylink W2.

I have no idea why some bolusses seem to take so long according to the log file. 
Also, I wonder: why doesn't this happen with SMB's, but only (in combination) with meal bolusses? 

Could it be, AAPS/Riley/pump is trying to handle multiple requests at the same time (e.g: SMB+meal bolus, or maybe even: meal bolus+pump sync)?

Why would it not be safe to delete bolusses, if they are not in the pump hisory after syncing?

andyrozman, 2022-06-11:
You can delete them manually... Situation that is happening with you is real anomally, which shouldn't happen... 

There are no multiple requests happening... AAPS has a Queue, so there wouldn't be two commands happening at same time. Unless you have connected with some other device to pump, which you shouldn't... AAPS needs to be the only device connecting to the pump...

If AAPS can't connect to RL then it would try to determine new connection speeds, in such case it could take a while. I will try to take a look at log to see what is happening...

old-square-eyes, 2022-06-11:
>  why doesn't this happen with SMB's, but only (in combination) with meal bolusses?

My example above shows a double SMB and a double wizard bolus.

andyrozman, 2022-06-11:
This is fix that might help. 

So when we do bolus on pump, before entry is written into database, timestamp is adjusted (it doesn't contain previous timestamp, when AAPS/orefx sent bolus command, but timestamp when bolus was received on pump), so when history comes back it will easily be tied to actual bolus record. This might prevent double bolus... 

Branch is: https://github.com/andyrozman/AndroidAPS/tree/bug1768_2xbolus

Milahan, 2022-06-12:
Thanks again, Andy! That sounds like a good solution.

I do have two questions, to verify if I completely understand what you meant.

You described the current AAPS-functionality as: "We pair boluses from pump history with local bolus, and difference can be only 1:59 minutes".
In the new branch you said in AAPS the "timestamp [bolus] is adjusted". 

Does this mean, the 2-minute-limit is gone? So AAPS will *always* match with the pump-history?

Did I understand correctly: will the 'bolus-time' in the treatments-tab be empty, until the pump confirms the bolus?

andyrozman, 2022-06-12:
No, limit is still there, but for example if AAPS starts bolus at 10:00 and queue is currently processing something, maybe Bolus would have to wait for few minutes so pump would get bolus at 10:04, previously we would still mark this bolus as 10:00 (as this is the time in the command), now this time will be adjusted so it will be have time 10:04. So when history comes back we can find which bolus from history is same as the one that was placed.. 

This already worked like this in 2.x, but with refactoring of whole driver (to kotlin and then adding new database stuff) this change seems to have goten lost...

Milahan, 2022-06-12:
Thanks for explaining and creating the fix! 
This really sounds like a solution.

andyrozman, 2022-06-12:
You misunderstood the "pairing" bit... When we place the bolus, we don't know what "id" will pump will assign, so we have to "pair" it after we get first reading of history. So when we place bolus in AAPS we store bolus with temporaryId and we also store that bolus in temporary storage, when we read history we then try to see which bolus from history is same as new bolus (in temporary storage), since bolus on pump can be also done manually, we need to find bolus that is nearest to the ones in history, which is why we look for 2 minutes difference. Time on pump is adjusted if difference is more than 20s, so such entry should be easily detectable...

andyrozman, 2022-06-12:
Opened PR: https://github.com/nightscout/AndroidAPS/pull/1825

closed 2022-06-12 by MilosKozak