nightscout/AndroidAPS #1771: [dev-l] After UI updtate graph labels doubled

swissalpine, 2022-05-26:
Milosz, thank you first of all for your tireless work.
Maybe the UI update is not ready yet yesterday. If it is, I would like to ask you to label again on the graph again (as before) only every 2nd line: So BG values only with 40-80-120-etc. and not show every hour twice.
This would be an optical advantage at least for small screens ...

[image]

swissalpine, 2022-05-26:
Addition: Compared to an old screenshot, it is noticeable that the number of grid lines has doubled, not the labeling. Maybe this is the way?

[image]

MilosKozak, 2022-05-26:
it can be related to graphview update. I didn't notice this change on my phone
Could you test reverting this commit https://github.com/nightscout/AndroidAPS/commit/4350a2d5d178038c1debe0e38556fb35fe8dd6d8 ?

osodebailar, 2022-05-26:
@MilosKozak they are (not) doubled, but it looks like than they are .
One hour has 2 entries: one for the full hour , one for the half hour . Maybe it’s configurable ?

swissalpine, 2022-05-26:
Thanks!
Reverting this commit fixes the issue.

While looking at the change before reverting I noticed that Android Studio showed me errors in the FixedLineGraphSeries.java file, e.g. there should be no (false) here:
```
            maxY = graphView.getSecondScale().getMaxY();
            minY = graphView.getSecondScale().getMinY();
```
And  the following line should overrides nothing:
```
@Override public void drawSelection(GraphView mGraphView, Canvas canvas, boolean b, DataPointInterface value) {}
 ```

MilosKozak, 2022-05-26:
these changes are related to change in lib

MilosKozak, 2022-05-26:
this is how i see latest dev
[image]

osodebailar, 2022-05-26:
This is my view :
[image]

MilosKozak, 2022-05-26:
I reverted the upgrade. I could not make it working like before.
I remember it was hard before too

closed 2022-05-26 by MilosKozak