nightscout/AndroidAPS #1780: Problem with autotune plugin

mordechi-david, 2022-05-29:
[autotune.zip](https://github.com/nightscout/AndroidAPS/files/8793398/autotune.zip)
The autotun plugin fails to compute for more than one day
And one day he fails most of the time only 3 times e
[AndroidAPS._2022-05-29_01-27-13_.56.zip](https://github.com/nightscout/AndroidAPS/files/8793400/AndroidAPS._2022-05-29_01-27-13_.56.zip)
[AndroidAPS._2022-05-29_01-27-13_.57.zip](https://github.com/nightscout/AndroidAPS/files/8793401/AndroidAPS._2022-05-29_01-27-13_.57.zip)

Philoul, 2022-05-29:
Can you disable "Include more log information for debugging" in settings and tell me if it's better ? (this setting should always is disabled)

mordechi-david, 2022-05-30:
Excellent !! Worked out, thank you very much, I did not see such a thing written in the release details.

closed 2022-05-30 by MilosKozak