nightscout/AndroidAPS #1781: AAPS shots down bc of pod fault 049.

Aliceladefoged, 2022-05-29:
Omnipod eros, orangelink, xdrip, nightscout. 

Pod exceeded 80 hours. For pod fault 028 and the aaps shot down every time I tried going to the omnipod page to change pod. 

Rebooted phone, uninstalled aaps, reinstalled, same error, nothing helps. 
Found out the imported settings remember the old pod. Imported older exported settings, and i could now go to the omnipod page. I discard the old pod. 
Tried to start a new post, but when it tries to connect to the new pod it gives me pod fault 049 again tells me to discard pod. It will not forget the old pod no matter what I do. And after 5 discarded pods in 2 hours of trying everything i am on the pdm now. 

This is so frustrating. 

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

nlof, 2022-08-12:
I had three pods shut down with 049 insulin delivery pod errors out of the last four pods. The first occured with v. 2.8.2.1, the following two were on a new phone, version 3.1.0.3.
This is the last log file and screenshots.
I posted the prior logs in a closed report, I'll look for it and post a link.

Milanakryskovskaya, 2024-04-23:
Error 049 is repeated regularly. Over the past month, 5 Omnipods have broken. The shelf life is good. Two Omnipods have broken down for the last two days in a row. AAPS 3.2.0.3.  I don't know what to do ((
Krutko5638@gmail.com
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/15082591/AndroidAPS.log)
[AndroidAPS._2024-04-22_06-16-16_.0.zip](https://github.com/nightscout/AndroidAPS/files/15082595/AndroidAPS._2024-04-22_06-16-16_.0.zip)

Milanakryskovskaya, 2024-04-23:
Error time:
23.04.2024   09:53
22.04.2024   05:48

samspycher, 2024-04-30:
Unfortunately it has been a long time since I have worked on this code, but I took another quick look.

Observations: as was mentioned, these errors are happening on the pod when temporary basals get requested from it.  From the log files, it isn't clear to me why the pods run into this type of error (I never worked on the pod messaging itself). Looking at the overall interaction across the logs I have looked at, a temp basal is sent before the one that errors out, and after the error happens, the pod is deactivated by AAPS.

@Aliceladefoged @nlof was there an outcome of this issue for you?

@Milanakryskovskaya here are two things that you perhaps can do/check:
- Perhaps you can collect temp basals that cause error 049 to see what is similar between them?
- Issue #384 had the 049 error code occur because of a time change that occured on the device during the preceding temp basal. You also have a preceding temp basal, but I don't see any time change in your logs, so your error 049 may not be time-related. Just to verify: do you know of anything time-change related that may be happening to your phone?

nlof, 2024-04-30:
@samspycher I had a few scattered errors with pods since, I don't remember what error codes though. From rereading my prior post, apparently I posted logs from 3 occurrences back then which could be used to investigate further...

samspycher, 2024-05-01:
@nlof thanks for your feedback. Yes, I checked your previous logs, they are all occuring on temp basals:
```
~/Downloads/nlof ls | while read datei; do echo $datei; cat "$datei" | egrep "Executing command:|Notification received: A Pod fault" | cut -c1-178 | egrep -v 'info.nightscout'; done
20220803-0120.log
11:39:16.953 [Thread-16160] D/PUMP: [OmnipodPumpPlugin.executeCommand():1106]: Executing command: [SET_TEMPORARY_BASAL]
11:39:21.707 [RxCachedThreadScheduler-2] D/NOTIFICATION: [NotificationStore.add():62]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
11:40:50.493 [Thread-16238] D/PUMP: [OmnipodPumpPlugin.executeCommand():1106]: Executing command: [GET_POD_STATUS]
11:40:54.093 [RxCachedThreadScheduler-2] D/NOTIFICATION: [NotificationStore.add():62]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
11:41:32.124 [Thread-16276] D/PUMP: [OmnipodPumpPlugin.executeCommand():1106]: Executing command: [DEACTIVATE_POD]
AndroidAPS (4).log
05:36:47.149 [Thread-28] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [SET_TEMPORARY_BASAL]
05:36:52.218 [RxCachedThreadScheduler-26] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
05:38:11.136 [Thread-9] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [GET_POD_STATUS]
05:38:14.918 [RxCachedThreadScheduler-25] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
05:38:14.927 [Thread-9] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [GET_POD_STATUS]
05:38:19.569 [RxCachedThreadScheduler-25] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
05:38:29.014 [Thread-15] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [GET_POD_STATUS]
05:38:32.284 [RxCachedThreadScheduler-25] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
05:52:22.557 [Thread-19] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [GET_POD_STATUS]
05:52:25.849 [RxCachedThreadScheduler-25] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
05:53:22.967 [Thread-12] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [GET_POD_STATUS]
05:53:27.561 [RxCachedThreadScheduler-27] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
05:53:27.581 [Thread-12] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [GET_POD_STATUS]
05:53:30.833 [RxCachedThreadScheduler-27] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
05:53:41.239 [Thread-14] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [GET_POD_STATUS]
05:53:45.288 [RxCachedThreadScheduler-27] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
05:54:04.864 [Thread-15] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [PLAY_TEST_BEEP]
05:54:10.240 [RxCachedThreadScheduler-27] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
05:58:08.849 [Thread-19] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [DEACTIVATE_POD]
AndroidAPS.log
00:06:37.210 [Thread-1266] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [SET_TEMPORARY_BASAL]
00:06:42.747 [RxCachedThreadScheduler-30] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
00:10:22.663 [Thread-1304] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [PLAY_TEST_BEEP]
00:10:26.596 [RxCachedThreadScheduler-30] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: A Pod fault (049 INSULIN_DELIVERY_COMMAND_ERROR) has been detected.
00:10:32.977 [Thread-1307] D/PUMP: [OmnipodErosPumpPlugin.executeCommand():1133]: Executing command: [DEACTIVATE_POD]
```

Given that I don't have much more to go on, the space of possible causes remains fairly large:
- A software or mechanical issue on the pod causes delivery to fail in certain rare cases (Omnipod, storage or age issues)
- Temp basal messages are faulty or are sent in a faulty manner that causes this error on the pod (AAPS)
- bluetooth stack / radio comms issues (phone manufacturer or Android)