nightscout/AndroidAPS #1789: Wear OS: Crash on scrolling menu settings

Andries-Smit, 2022-05-30:
With the latest build 3.0.0.2dev-l On the watch app
- double click on a watch face to open the menu 
- got to Settings 
- Scrolling down the app 
- Result forces the app to close

With an exception at java.io.FileNotFoundException: /storage/emulated/0/AAPS/logs/info.nightscout.androidaps/AndroidAPS.log: open failed: ENOENT (No such file or directory)

**The ADB log shows**
2022-05-30 20:51:45.815 592-690/? I/InputReader: Touch event's action is 0x1 (id=5, t=0) [pCnt=1, s=] when=880203.573598
2022-05-30 20:51:45.938 1199-25003/? E/MediaProvider: Creating a non-default top level directory or deleting an existing one is not allowed!
2022-05-30 20:51:45.941 1199-8809/? E/MediaProvider: Creating a non-default top level directory or deleting an existing one is not allowed!
2022-05-30 20:51:45.978 25525-25525/? I/System.out: 20:51:45,673 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Found resource [assets/logback.xml] at [assets/logback.xml]
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,691 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - debug attribute not set
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,823 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - About to instantiate appender of type [ch.qos.logback.core.rolling.RollingFileAppender]
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,824 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - Naming appender as [file]
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,876 |-INFO in c.q.l.core.rolling.TimeBasedRollingPolicy - Will use zip compression
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,878 |-INFO in c.q.l.core.rolling.TimeBasedRollingPolicy - Will use the pattern /storage/emulated/0/AAPS/logs/info.nightscout.androidaps/AndroidAPS._%d{yyyy-MM-dd}_%d{HH-mm-ss, aux}_.%i for the active file
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,881 |-INFO in ch.qos.logback.core.rolling.SizeAndTimeBasedFNATP@99e5b03 - The date pattern is 'yyyy-MM-dd' from file name pattern '/storage/emulated/0/AAPS/logs/info.nightscout.androidaps/AndroidAPS._%d{yyyy-MM-dd}_%d{HH-mm-ss, aux}_.%i.zip'.
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,881 |-INFO in ch.qos.logback.core.rolling.SizeAndTimeBasedFNATP@99e5b03 - Roll-over at midnight.
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,886 |-INFO in ch.qos.logback.core.rolling.SizeAndTimeBasedFNATP@99e5b03 - Setting initial period to Mon May 30 20:51:45 GMT+02:00 2022
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,892 |-INFO in ch.qos.logback.core.joran.action.NestedComplexPropertyIA - Assuming default type [ch.qos.logback.classic.encoder.PatternLayoutEncoder] for [encoder] property
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,930 |-INFO in ch.qos.logback.core.rolling.RollingFileAppender[file] - Active log file name: /storage/emulated/0/AAPS/logs/info.nightscout.androidaps/AndroidAPS.log
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,930 |-INFO in ch.qos.logback.core.rolling.RollingFileAppender[file] - File property is set to [/storage/emulated/0/AAPS/logs/info.nightscout.androidaps/AndroidAPS.log]
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,943 |-ERROR in ch.qos.logback.core.rolling.RollingFileAppender[file] - Failed to create parent directories for [/storage/emulated/0/AAPS/logs/info.nightscout.androidaps/AndroidAPS.log]
2022-05-30 20:51:45.979 25525-25525/? I/System.out: 20:51:45,946 |-ERROR in ch.qos.logback.core.rolling.RollingFileAppender[file] - openFile(/storage/emulated/0/AAPS/logs/info.nightscout.androidaps/AndroidAPS.log,true) failed java.io.FileNotFoundException: /storage/emulated/0/AAPS/logs/info.nightscout.androidaps/AndroidAPS.log: open failed: ENOENT (No such file or directory)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at java.io.FileNotFoundException: /storage/emulated/0/AAPS/logs/info.nightscout.androidaps/AndroidAPS.log: open failed: ENOENT (No such file or directory)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at libcore.io.IoBridge.open(IoBridge.java:492)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at java.io.FileOutputStream.<init>(FileOutputStream.java:236)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.core.recovery.ResilientFileOutputStream.<init>(Unknown Source:7)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.core.FileAppender.openFile(Unknown Source:57)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.core.FileAppender.start(Unknown Source:58)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.core.rolling.RollingFileAppender.start(Unknown Source:216)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.core.joran.action.AppenderAction.end(Unknown Source:11)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.core.joran.spi.Interpreter.callEndAction(Unknown Source:23)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.core.joran.spi.Interpreter.endElement(Unknown Source:32)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.core.joran.spi.Interpreter.endElement(Unknown Source:11)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.core.joran.spi.EventPlayer.play(Unknown Source:84)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.core.joran.GenericConfigurator.doConfigure(Unknown Source:16)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.core.joran.GenericConfigurator.doConfigure(Unknown Source:34)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.core.joran.GenericConfigurator.doConfigure(Unknown Source:7)
2022-05-30 20:51:45.979 25525-25525/? I/System.out:     at 	at ch.qos.logback.classic.util.ContextInitializer.autoConfig(Unknown Source:36)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at org.slf4j.impl.StaticLoggerBinder.init(Unknown Source:7)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at org.slf4j.impl.StaticLoggerBinder.<clinit>(Unknown Source:16)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at org.slf4j.impl.StaticLoggerBinder.getSingleton(Unknown Source:0)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at org.slf4j.LoggerFactory.bind(LoggerFactory.java:150)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at org.slf4j.LoggerFactory.performInitialization(LoggerFactory.java:124)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at org.slf4j.LoggerFactory.getILoggerFactory(LoggerFactory.java:417)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:362)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at info.nightscout.shared.logging.AAPSLoggerProduction.debug(AAPSLoggerProduction.kt:22)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at info.nightscout.androidaps.Aaps.onCreate(Aaps.kt:27)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1192)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6746)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at android.app.ActivityThread.access$1300(ActivityThread.java:241)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1920)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at android.os.Handler.dispatchMessage(Handler.java:106)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at android.os.Looper.loop(Looper.java:246)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at android.app.ActivityThread.main(ActivityThread.java:7690)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at java.lang.reflect.Method.invoke(Native Method)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:593)
2022-05-30 20:51:45.980 25525-25525/? I/System.out:     at 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:995)
[... cut ...]
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-05-31:
this is not related to any crash ... only missing permission on watch. 
i don't think it's necessary to store logs on watch because of it's battery and memory

reijoahola-nightscout, 2022-05-31:
I tested this with the latest 3.0.0.2-dev-l (a091b6c32) and using TicWatch Pro (year 2018 model). I did not have the same problem, I could easily scroll the settings menu to the bottom in the watchface (AAPS Digital Style).

Andries-Smit, 2022-06-01:
After uninstalling and installing again, with default settings,  it does not crash. So it could be a non-default setting or an error in the stored settings.

I would agree that logs should not be enabled, by default. maybe it should be a switch users can enable in case they have troubles

closed 2022-06-02 by MilosKozak