nightscout/AndroidAPS #1791: BYODA calibration =》red triangle in AAPS

robertrub, 2022-05-30:
@MilosKozak I got the difficult to catch calibrating in BYODA giving a red triangle bug once more. I hope the logs will help you this time.

Around 22h27, I calibrate in BYODA. It was showing sth like 190 and calibration was higher, sth like 215... Next BG read, BYODA and xDrip shows 216 but AAPS shows 195. I look in BYODA tab in AAPS, the time stamp is correct 22:27 but BG is not 216 as in xDrip and BYODA.

5 min later, the BGs aligne on the 3 aps but I get the red triangle in AAPS.

[image]

[image]

[image]

[image]

And the logs. Look a bit before and after 22:27...

[image]

MilosKozak, 2022-05-31:
this is impossible to detect from received batch (only 1 wrong value). The only way would be compare with database which would be expensive
I'll keep this open to see if it's rare or not

robertrub, 2022-06-01:
Ok. It is in fact rare as I didn't have this bug for over a month (and I calibrated around the 5 min delay often).

Will come back if it happens again.

vanelsberg, 2022-06-06:
>  in fact rare

Interesting. 
Calibrating G6/BUODA often (avg once every 2 days). Haven't seen any red triangles for a long time ...

vanelsberg, 2022-06-13:
Weird problem: Have seen this in the past (weeks ago).
I cannot repro this. Tried several times.
Also I calibrate my G6 every 2 days on average. So I should see this at least occasionally

robertrub, 2022-06-18:
Hi @MilosKozak Got the red triangle once more and this time I was about 3 min after the sensor BG measure. The difference between sensor and blood was important (G6 54 vs blood 70).

[image]

[image]

[image]

I hope you'll get more info from these logs.

[image]

robertrub, 2022-07-13:
@MilosKozak I don't know if you did any work on the subject. I calibrated at 7:39 just after the 5 min top at 7:38.

There is a slide on the next BG in AAPS 7:44 that does not respect the 5 min intervals but had no flags.

To make it short : all is working ok.

[image]

[image]

After 5 tries, I can't upload a logs :( The GitHub app needs debugging too :) Will try later

robertrub, 2022-07-13:
Ok, after 10 trials I give up. Tell me if you need the logs...

closed 2022-07-13 by MilosKozak