nightscout/AndroidAPS #1792: Can't stop "pump unreachable" alarm

robertrub, 2022-05-31:
This had happened before several weeks ago. I get a "pump unreachable" alarm beeps in the middle of the night. Half awake, I hit "Ok" and do a manual refresh.

Then the alarms starts again. I don't see any window with buttons to stop it. I swipe all the AAPS messages in Android (still half awake and without glasses)... But the alarm goes on. Refresh pod status does not change anything.

3 dots menu, quit AAPS. Alarm stops but AAPS restarts by itself with and the alarm too, except that the sound is different.

I had to force stop AAPS and empty cache memory to stop the alarm.

Logs at around 02:13

[image]

[image]

MilosKozak, 2022-06-08:
i improved logging in this area https://github.com/nightscout/AndroidAPS/commit/024a90231641c108e4aa82776ed2f501e8e7a50a
provide new logs when this happen again

vanelsberg, 2022-06-13:
Comfirming, happened to me some build ago. One time only, was unable to repro.
Will provide logs when this happens ...

robertrub, 2022-06-29:
@MilosKozak Happened again with last dev n.

[image]

Phone near me, I'm probably sleeping on my arm on the pod. Pod unreachable alarm. I mute or OK (not sure as I don't have my glasses and just woke up). I actualized Dash pod and it worked ok. Then, it looked as if AAPS had just restarted : no more predictions and the progression bar at work. The unreachable alert sound was there again but no message and no way to turn it off.

I quit AAPS via the 3 dots menu. As usual AAPS starts again by itself. This time, the alarm sound is still there but the sound has changed (a kind of blip) but there still is no message to turn off the alarm.

I had to kill AAPS and clear cache in order to stop the alert.

Logs attached. Look a bit before 5AM.

[image]

robertrub, 2022-07-18:
It happened again last night around 0h20. Had to restart phone. I hope the logs did not dispapear with the reboot...

AndroidAPS 3.0.0.2-dev-o
Build: e3339d8a5-2022.07.17-16:37
Remote: https://github.com/nightscout/AndroidAPS.git
Flavor: fullrelease
Version Nightscout : 14.2.5
Mode ingénierie actif

[AndroidAPS_LOG_1658096638492.log-1.zip](https://github.com/nightscout/AndroidAPS/files/9130655/AndroidAPS_LOG_1658096638492.log-1.zip)

MilosKozak, 2022-07-18:
thanks.
it seems to be only concurrecy issue...
I made some improvements to dev. Report new logs if the issue re-appear

robertrub, 2022-11-02:
Hi

It happened again. G6 change. About 10 min after the BGs start to come in, AAPS rings the alert but no button to turn it off. Actualize pod does not cut the alarm.

I had to quit using the 3 dots menu and the alert stopped immediately.

[image]

Will add logs. Look at around 14:08

robertrub, 2022-12-31:
Happened again in dev-f at 9AM. I was near the phone and doing nothing on the phone. The alarm started, no messages, beep pod worked ok but didn't stop the alarm. Actualize did not stop it either. I had to stop AAPS to the alarm.

[image]

Harrypatz, 2023-02-01:
It's been happening since 2.7, I've reported it before, and not alone. Have had 4 different smartphones since then and it keeps coming back. Mostly when you can't use it, in meetings or at work or at night. I'm already thinking about getting an old smartphone and working with 2.6 again

MilosKozak, 2023-08-17:
Closed for inactivity. Create new issue is if you still see the problem on latest branch

closed 2023-08-17 by MilosKozak