nightscout/AndroidAPS #1805: 3.0.0.2-dev-l: mmol/l / mg/dl bug

diabetesdiyaps, 2022-06-04:
Build version: 3.0.0.2-dev-l

Yesterday setup a test environment for dynISF using my phone and personal accounts (Nightscout, Heroku, xDrip) for it. All have the same version as my girlfriend. My T1D girlfriend is using the production environment ;-) 

Found a bug in dev-l in conversion mmol/l and mg/dl: 

- We use mmol/L settings, but when using bolus calculator, 'blood glucose' value shows in Nightscout (most recent v14.2.5) report Treatments as mg/dl instead of mmol/l. After reverting back to 3.0.0.2 and using bolus calculator the same I see 'blood glucose' value is stored again as mmol/l.

- Also, after using AndroidAPS bolus calculator entering value in mmol/l the overview screen suddenly shows the value as mg/dl. When changing AndroidAPS unit setting from mmol/l to mg/ml overview screen seems to show to correct value.

See screenshots for a deep dive: 

[image]
With dev-l in main window when I switch units it goes wrong:

[image]
Set AndroidAPS units in mmol/l

[image]
Set AndroidAPS units in mg/dl

[image]
No stubs in this test-round ;) 

For my the upcoming days / weeks it's possible to re-test, upload log files or build a new dev version to test. Love to help. 

Dev's, great job so far !!

All the best,
Peter

petervanrijt, 2022-06-04:
Created this issue this afternoon with my testing account. Response using my personal account. Did some testing tonight:

### AndroidAPS 3.0.0.2

1. AndroidAPS in mmol/l, Nightscout in mmol/l. Bolus wizard BG 5.5, carbs 20g. In Nightscout report treatments: BG 5.5
2. AndroidAPS in mmol/l, Nightscout in mg/dl . Bolus wizard BG 5.5, carbs 20g. In Nightscout report treatments: BG 5.5
3. AndroidAPS in mg/dl , Nightscout in mmol/l. Bolus wizard BG  99, carbs 20g. In Nightscout report treatments: BG 99
4. AndroidAPS in mg/dl , Nightscout in mg/dl . Bolus wizard BG  99, carbs 20g. In Nightscout report treatments: BG 99

### AndroidAPS 3.0.0.2-dev-l

5. AndroidAPS in mmol/l, Nightscout in mmol/l. Bolus wizard BG 5.5, carbs 40g. In Nightscout report treatments: BG 99
6. AndroidAPS in mmol/l, Nightscout in mg/dl . Bolus wizard BG 5.5, carbs 40g. In Nightscout report treatments: BG 99
7. AndroidAPS in mg/dl , Nightscout in mmol/l. Bolus wizard BG  99, carbs 40g. In Nightscout report treatments: BG 99
8. AndroidAPS in mg/dl , Nightscout in mg/dl . Bolus wizard BG  99, carbs 40g. In Nightscout report treatments: BG 99

Expected in 5 and 6 was a BG in Nightscout report treatments of 5.5 instead of 99.

petervanrijt, 2022-06-06:
Hi, 

An update on the findings. Writing to NIghtscout still is in mg/dl instead of mmol/l. But, in a chat with another Dutch user I heard he did not have trouble with his screen "mmol/l / mg/dl stripes" to occur. So I have tested again. 

I saw when the switch is "too fast" there's a screen error (see below). When I don't wait for the blue bar to completely finished to right the screen has "mmol/l mg/dl stripes". A good workaround is to Exit the app and restart, the screen will be OK after loading. 

Perhaps an idea to pop-up a message after switching from mg/dl to mmol/l to wait completely for the screen to load, or to Exit the app and reboot immediately?

Thanks and all the best,
Peter

[image]

MilosKozak, 2022-06-08:
issue is on NS side. it doesn't convert value to units selectwed in settings.
I set at least upload to units selected in AAPS https://github.com/nightscout/AndroidAPS/commit/70f365960052ec28f6844538559c8cd38bf32220
So far it was always in MGDL

closed 2022-06-08 by MilosKozak

petervanrijt, 2022-06-10:
Thank you @MilosKozak !