nightscout/AndroidAPS #1821: Suggestion for smb improvement

mordechi-david, 2022-06-10:
I accompany dozens of loops in Israel and I noticed something that would greatly improve the accuracy of smb handling,
There is a problem when setting a maximum of minutes to smb at certain hours it should be much more aggressive and although it is by basal it is still inaccurate and the result (especially in children) is that if you set a relatively high number of minutes it brings some hours to hippo and if you lower the number of minutes it does not Enough facility at other times which is very important,
As a pirated solution I set in these hours a higher basal than needed to allow the smb to work harder also at certain hours I wanted there to be no smb at all I automated a temporary target value 101 for those hours and set the smb not to work with high target values ​​and so it works great
But there is a possibility to greatly improve everything in a much simpler way, just put in the profile settings an additional value of smb limit that can be changed by the hours of the day and then it is possible even at certain hours to set 0 and it will disable smb in those hours
In my experience this will be a huge improvement to the ability to use smb especially in children
Thanks for all the amazing work

koelewij, 2022-06-13:
It would definitely not hurt to have this capability in the profile and allowing to set it for different time ranges.
People can just leave it to one value if they would prefer that.

Andries-Smit, 2022-06-27:
@mordechi-david do you have some suggestions to improve Right-to-Left languages? https://www.facebook.com/groups/AndroidAPSUsers/posts/3307543259467068

yhonatanb20, 2022-07-18:
I disagree, there are some issues, ex. deltha displayed inproperly and arrow comes before BG value:
[image]
Also the ex. 4h3' configuration of indicating TT/Profile duration doesn't quite work on hebrew (something like h43' displayed).
Overally the right-to-left support is good, but I wished I could use AAPS in English while my phone is set to hebrew, which doesn't work on 3.0.0.1 but seems to work on 3.1.0-beta1, hope it will work on 3.0.1

FlyRad, 2022-07-29:
I also think it would be a good feature to blog SMBs in ceartain times. I also found a solution that works for me but it is not very practical. I raised the ISFs dramatically in this times where I do not want that SMBs are given....mostly at the end of a still working eating bolus when glucose reaches the aim (end of DIA). Especially in this timezone a SMB can bring you into Hypo easily. Changing ISFs or blogging SMBs in certsin times has a adisadvantage. If you change eating hours the changed ISFs or blogged SMBs will not fit to the meal anymore!
I think the times for blogging SMBs should be connected to the times when you give youre bolus for your meal. e.g. After mealbolus was given blog SMB in hour 5 and 6 afterwards. Another idea would be that blogging duration raises with the amout of insulin that was given for the meal. I know it is complecated but an extra idea🙂