nightscout/AndroidAPS #1823: New IOB is not displayed immediately if an occlusion occurs.

Korostensky, 2022-06-11:
Reporting bugs
--------------
- **09:36:13.475 Error detected: Okluze** 
-  The history from the pump (IOB,COB) has not changed (were not in treatment in NS) if an occlusion occurs (Asked: 4.5 Delivered: 2.6). 
- It was displayed after the next bolus with the new values (both of them IOB +2.6+2.9, COB +35+30). I cancelled the second bolus (Asked: 3.85 Delivered: 2.9) because I was counting on some insulin already dispensed the first time, but I didn't see it. 
- Maybe it's a behavior not a mistake. If the bolus fails, the carbs don't register. They do load the next time the pump is reconnected, but the insulin dosed is something else. 
- Maybe waiting is enough, but I think the history from the pump should be loaded and displayed immediately after occlusion, that's potentially dangerous.
- 3.0.0.2-dev-m-fa79864d2-2022.06.08-13:36, 
- DANA_RV2
- [AndroidAPS_LOG_1654933557660.log.zip](https://github.com/nightscout/AndroidAPS/files/8883899/AndroidAPS_LOG_1654933557660.log.zip)

Korostensky, 2022-06-13:
The next day 2022-06-12, same situation. Occlusion showed information about incomplete bolus, but IOB and COB did not change. 

This time the app got stuck and only after a restart did the IOB and COB change
[AndroidAPS_LOG_1655018663768.log.zip](https://github.com/nightscout/AndroidAPS/files/8888294/AndroidAPS_LOG_1655018663768.log.zip)

MilosKozak, 2022-06-13:
hmmm pump stops communicating on this event and it takes time to recover from this situation. Anyway this https://github.com/nightscout/AndroidAPS/commit/f887043aff9abccf3106112af667669e21e374b2  could speed it up

closed 2022-06-13 by MilosKozak