nightscout/AndroidAPS #1833: Dashboard minor bugs 3.0.0.2-dev-m

Korostensky, 2022-06-15:
Reporting bugs on dashboard
"3.0.0.2-dev-m"
--------------

# 1. When you remove an action from the treatment agenda, the information does not disappear from the chart (only not deleted).

[image]

[image]

# 2. If I start a new sensor (Dexcom G6), the system calculates the values and displays a warning. But this is displayed for more than one day (no longer shown in the graph, it's confusing) The exclamation mark symbol should disappear when you click "OK" or when the recalculation is not displayed.

[image]
[image]

MilosKozak, 2022-06-15:
1) these are announcements - must be deleted on 5th tab
2) icon doesn't disappear after click. while using Dexcom you can see exclamation mark only on transmitter change for about day and half. If it's different situation, your configuration is somewhere wrong and there are doubled data

Korostensky, 2022-06-15:
Thank you for the explanation, I agree and apologize for the unnecessary comments.

1. So my deletion only modified the IOB/COB and not the treatment, so it must be deleted from two places. 
But this is very handy (the app doesn't calculate the values, but I can see what I was doing for the changes)

2. I really changed the transmitter, a day and a half is fine.

closed 2022-06-15 by Korostensky