nightscout/AndroidAPS #1852: Users need a way to archive objectives achieved, independently from preferences

vanelsberg, 2022-06-25:
APS version 3.0

Users need a way to archive objectives achieved, independently to preferences.

As is now, restoring objectives can only be done by way of importing an old settings file.
Preferences from the old settings file are most likely outdated and will overwrite current settings.

vanelsberg, 2022-06-25:
Possible solution:
While importing give the user the option to import "Objectives only"?

MilosKozak, 2022-06-25:
80% of users would do this instead of pass them

closed 2022-06-25 by MilosKozak

vanelsberg, 2022-06-26:
> 80% of users would do this instead of pass them

@MilosKozak Not sure what you are saying?
Objectives export/import is (should be) encrypted in the same way preferences are?

MilosKozak, 2022-06-26:
everyone would look for any export on internet and just import finished preferences