nightscout/AndroidAPS #1853: Omnipod EROS: There is no way to forcefully wipe/reset the pod state in case of "emergency"

vanelsberg, 2022-06-25:
AAPS version 3.0.0.1 release and higher
Driver: Omnipod EROS

Problem description:
The Omnipod EROS driver has a bug that can lead to a situation where users get stuck with a pod registration that can not be "forcefully" discarded. Several users reported running into this problem as a result of Pod failure.

Work around used for not losing Objectives is importing 3.0 settings in AAPS version 2.8.1, discard the Pod and then export preferences that can be imported by AAPS 3.0

Proposed work around
Until the actual bug is solved, but also in general, I think the EROS driver should have a way to "wipe" the Pod state in case of "emergencies". Up to AAPS version 2.8 this could be done by editing the unencrypted preference file.
As of version 3.0 this is no longer an option.
How about an option to clear Pod state when running in developer mode?
This way users have an "escape" option to get out of situations where there is Pod state without option to Discard.
For AAPS 2.8 editing the preference file was used as a last resort

vanelsberg, 2022-06-25:
Alternative solution to the problem would be resetting AAPS preferences or a reinstall.
An option to import "objectives only" would then ensure objectives are not lost

https://github.com/nightscout/AndroidAPS/issues/1852

vanelsberg, 2022-06-25:
Remark:
Omnipod DASH implementation ensures that after 3 unsuccesfull retries user is offered the option to "forcefully" discarding the Pod.

Something like this maybe could also be implemented for EROS but expect this would need non-trivial driver changes?
Expecting forcefully wiping pod state can be done with minimal impact on drive code?

MilosKozak, 2022-06-25:
what keys do you have to remove?

vanelsberg, 2022-06-25:
> what keys do you have to remove?

@MilosKozak 
Clearing `content: "AAPS.Omnipod.pod_state"` should effectively remove the Pod.

Currently not using EROS so unable to validate. Do have 3 EROS pods reserved for dev, so can test when needed.

vanelsberg, 2022-06-25:
Hint:
Think PodState is accessible at `java/info/nightscout/androidaps/plugins/pump/omnipod/eros/manager/AapsErosPodStateManager.java`
See methods storePodState() and readPodState()?

vanelsberg, 2022-06-25:
[image]

MilosKozak, 2022-06-25:
Discard pod is not enough in this state?

vanelsberg, 2022-06-25:
The bug/problem is that AAPS/EROS somehow gets in a state where activation unexpectedly failed and activation process can not be continued. In that state however the "Discard" button is still Disabled (it believe it is only enabled after successful activation)

vanelsberg, 2022-06-25:
Most obvious solution would be to target this specific bug (proper error handling?). But that would mean touching driver code which you may not want at this stage?

Also in any other situation, there is still no ways to repair by wiping state?

MilosKozak, 2022-06-25:
https://github.com/nightscout/AndroidAPS/commit/00f0446c21e059f1287333fc72578daeb6f8c8c2

vanelsberg, 2022-06-25:
> [00f0446](https://github.com/nightscout/AndroidAPS/commit/00f0446c21e059f1287333fc72578daeb6f8c8c2)

Looks good. However not sure if this would do the trick (had been looking at something like this)
Problem is I suspect podStateManager.isPodInitialized to evaluate to true in this case (as Pod activation is started but ongoing)?

vanelsberg, 2022-06-25:
Nice "solve". Maybe add this to dev and let "testers" look at this.
(Testing this: Of only I new a way to reproduce the originating problem...)

vanelsberg, 2022-06-25:
Will ask around for a settings file that has the problem so I can test...

vanelsberg, 2022-06-25:
Thanks @MilosKozak for taking this so fast!

MilosKozak, 2022-06-25:
you can test it just before pod expiration (to not waste it)

vanelsberg, 2022-06-25:
Thanks. That is the safest way....
But not needed. Discard (should only) remove the pod registration without actually de-activating the Pod.
So, after Discarding it should be recoverable through importing settings (which contains the pod stat)

vincent1890, 2022-07-02:
Hello,
Unfortunately I use the dev version and the delete button does not work tested on two different pod (see screenshot
[image]
[image]
)
I have been forced to re-import a backup without pod to be able to use a new pod without failure

vanelsberg, 2022-07-02:
> ... Unfortunately I use the dev version and the delete button does not work ...

@MilosKozak 
Confirming. The "Discard" button is active and responsive but pressing it does not have the expected effect (no errors)
Assuming the underlying code does not allow discarding while the pump is in this state....

vanelsberg, 2022-07-02:
@MilosKozak Should we re-open this issue or do new?

MilosKozak, 2022-07-02:
I need more info what must be cleared

happygirlnyc70, 2022-07-02:
I can confirm this as well.  I have used both 3.1 beta and dev-n 4f0f01883 with the same result.

vincent1890, 2022-07-03:
@MilosKozak 
> I need more info what must be cleared

Hello
I don’t know if my answer will bring an element of answer but it would have to go back to the state when a POD is deactivated correctly "No active POD".
Currently dasn the capture shown have little to see that a POD is still half active in the AAPS system

vanelsberg, 2022-07-03:
> I need more info what must be cleared

@MilosKozak Discarding Pod is not Discarding properly....
Think I found the (underlying) problem is in not discarding/handling pod state correctly. Have to look into this ...
Will do a PR this week (currently family matters take prio).

vanelsberg, 2022-07-04:
@MilosKozak 
Preparing a fix though https://github.com/nightscout/AndroidAPS/issues/1882

vanelsberg, 2022-07-09:
Expect problem to be solved through https://github.com/nightscout/AndroidAPS/issues/1882

closed 2022-07-09 by MilosKozak