nightscout/AndroidAPS #1857: Settings lost after phone hard reboot

bchris21, 2022-06-26:
Version: Dev-n
Android Version: 9

Phone was squeezed in my bag and power button was pressed. Phone hard rebooted 3-4 times.

When realized AAPS was on open loop. After running loop I got the message that I had to pass objective 1.

Reverted on previous good settings (a month ago) and as expected Pod was lost.

Reverted back to Dev-m and started a new Pod.

Strange thing is that I didn't have to discard the old Pod. I just started a new one.

Logs attached. Issue happened around 15:45 local time.

[image]

[image]

[image]

MilosKozak, 2022-06-26:
that's something I cannot affect

bchris21, 2022-06-26:
Ok maybe it's phone related.

A bit more info:

- Papertrail data transfer limit reached.
- There is a slow DBUPDATE of all BGs in NSClient.

Can I assume that the whole database was pruned?

Thanks

MilosKozak, 2022-06-27:
local storage is android thing. AAPS can only use it but not reset it itself

closed 2022-06-27 by MilosKozak