nightscout/AndroidAPS #1858: Profile shift can lead to missing profile values

justmara, 2022-06-26:
Having these profile settings
[image]

If you do profile shift for 2 hours, you'll get target of 5 for the whole day since 9:00
[image]
[image]

The second range of '02:00 to 09:00' disappeared, but its value moved to (used to be) third range of `09:00...`

AndroidAPS 3.0.0.2 built from master.

MilosKozak, 2022-06-27:
can you test the same with beta1 ? i already checked this and it was working properly

justmara, 2022-06-28:
only could check 3.1-beta in emulator - works fine there.

MilosKozak, 2022-06-28:
ok, it will be released soon

closed 2022-06-28 by MilosKozak