nightscout/AndroidAPS #1866: Missing blood glucose alarm

n0rb33r7, 2022-06-29:
ADD TIME OF EVENT HERE: 15.57
ADD ISSUE DESCRIPTION: Missing blood glucose alarm at 15.57 CEST, it got fixed on its own a minute later, but I could see that it was behind of ~1500 minutes
-------------------------------------------------------
(Please remember this will send only very recent logs.
If you want to provide logs for event older than a few hours,
you have to do it manually)
-------------------------------------------------------
AndroidAPS 3.0.0.2-dev-n
Build: 9b6477381-2022.06.29-10:06
Remote: https://github.com/nightscout/androidaps
Flavor: fullrelease
Nightscout version: 14.2.5
Engineering mode enabled

MilosKozak, 2022-06-29:
i don't fully understand

n0rb33r7, 2022-06-30:
Hi Milos,

At 15:57, I've got an alarm in AAPS that I'm having "Missed BG readings":

15:57:52.682 [NSClientServiceHandler] D/EVENTS: [RxBus.send():23]: Sending EventNSClientNewLog[action=DBADD treatments,date=1656511072681,logText=Sent PairTherapyEvent {"eventType":"Announcement","isValid":true,"created_at":"2022-06-29T13:56:26.371Z","enteredBy":"AndroidAPS","units":"mg\/dl","notes":"Missed BG readings","isAnnouncement":true} 476/477,timeFormat=java.text.SimpleDateFormat@8140d380]

When I had the "Missed BG readings" alarm, I've checked AAPS and it was showing a Sensor Glucose value 1550 minutes before

By 15:58, it was showing the actual sensor glucose values

MilosKozak, 2022-10-04:
unknown reason

closed 2022-10-04 by MilosKozak