nightscout/AndroidAPS #1871: 3.1beta Watchface isnt connecting.

FreDiabetics, 2022-06-30:
I'm using a GalaxyWatch4 series and updated to the 3.1beta. No watchface is connecting. Even if I press send values to watch again nothing happens. The aaps complictaions showing !err!
I deleted the wear apk from the watch before I installed the 3.1beta wear apk.

[image]
[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-06-30:
did you build new wear apk?

FreDiabetics, 2022-07-01:
> did you build new wear apk?

yes I builded the new wear apk and sende it to the watch with android studio. I deleted the old wear apk before

[image]
.

FreDiabetics, 2022-07-05:
is there any tip or workaround?

CarnageMarkus, 2022-10-01:
Same issue here

MilosKozak, 2023-08-17:
Closed for inactivity. Create new issue is if you still see the problem on latest branch

closed 2023-08-17 by MilosKozak