nightscout/AndroidAPS #1875: 3.1-beta1 Calibration button missing

PetrMlejnek, 2022-07-01:
I'm missing Calibration button that should send me to BYODA app to calibrate a sensor.
See: https://www.youtube.com/watch?v=bwGZnAQvXGQ

Have this problem since AAPS version 3 (3.0.0.1, 3.0.0.2, 3.1-beta1)
BYODA version: 1.9.0.5, 1.10.0.2
Last AAPS version without this problem was 2.8.2

[image]

MilosKozak, 2022-07-01:
for BOYDA it's not working properly thus feature has been removed

closed 2022-07-01 by MilosKozak

PetrMlejnek, 2022-07-03:
> for BOYDA it's not working properly thus feature has been removed

It's a shame, I personally used it every day (1-2x), because I have to calibrate my sensors (even Dexcom G6).