nightscout/AndroidAPS #1876: Rileylink is not connecting

toniteh78, 2022-07-01:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

toniteh78, 2022-07-01:
[image]

MilosKozak, 2022-07-01:
fixed in dev

closed 2022-07-01 by MilosKozak