nightscout/AndroidAPS #1877: In Bolus calculator the insulin coloring in dark blue is hard to read

AnagarsWelt, 2022-07-03:
if you have a dark background, then the blue of the insulin value is hard to read for visually impaired users. On the overview the dark blue is visible, since there is no text. Suggestion: print the insulin to be delivered in white - even though it is not correct in the color scheme. The number should be a high-contrast, since it is to important a value to miss interpret. 
Build: 37aebc157-2022.06.26-09:37

osodebailar, 2022-07-03:
To verify better , please send a screenshot

osodebailar, 2022-07-03:
Do you mean the widget ?

AnagarsWelt, 2022-07-03:
I mein the IE in blue from the Calculator.

Alamo04, 2022-07-04:
> Do you mean the widget ?

Widget has nothing to do with calculator. 
His problem is to read the sugguetion in the blue color of the calculator.

osodebailar, 2022-07-04:
For me the contrast is high enough .
What do you think @MilosKozak ?
The only possible suggestion is to make the font in bold .

Andries-Smit, 2022-07-04:
The accessibility scanner https://play.google.com/store/apps/details?id=com.google.android.apps.accessibility.auditor&hl=en&gl=US
Is a great tool to check contrast. There are many suggestions for improvements, but the contrast of the blue suggestion is sufficient.

AnagarsWelt, 2022-07-04:
Sorry, but the contrast between blue and Black is Not enough for perple wo can'see so good. Rally!!!!!!

osodebailar, 2022-07-04:
Probably it can be a solution to make the font bold.

Andries-Smit, 2022-07-04:
@AnagarsWelt (please keep the discussion polite. No need to shout with exclanation in a discussion. Please note, in English exclamations !!! are only used when you are angry. I guess you are not. Everyone is here to help without trying to insult)

Anyway, everyone is different, (cultural and) visual capabilities. The app is implemented with the recommended contrast ratio, to support most people.
Are there more places where in the app where the contrast is not sufficient?

Also, have a look, Android also have accessibility features to improve contrast, you can play with them to see if this helps you with contrast issues in general.  https://support.google.com/accessibility/android/answer/11183305?hl=en#highcontrast

AnagarsWelt, 2022-07-04:
Erst einmal entschuldige ich mich, ich wollte keinesfalls unhöflich sein- nur auf die hohe Bedeutung hinweisen.

Nun muss ich auf Deutsch schreiben, da es zu kompliziert für mich auf Englisch wird. Hoffentlich ist das in Ordnungso.

 

Vielen Dank für die Hinweise, um selbst Änderungen vornehmen zu können-mir geht es aber lediglich um den beschriebenen blauen IE Wert, denn mit dem Rest bin ich sehr glücklich.

Da der Untergrund des  Dialogfeld des Bolusrechners in mittelgrau gehalten ist, überstrahlt dieses den kleinen rechteckigen schwarzen Hintergrund, der unter dem blauen IE  ( = errechneten Insulins ) Feld  ist. Das ist leider für viele Seheinschränkungen so. Fett wäre sicher besser- am Besten aber weiß. Das es größer als alle anderen Zahlen ist, ist auch sehr schön.

Macht man das Dialogfeld  „ Bolus“, „ Insulin“,“ Kohlehydrate“ auf, sind  alle Zahlen  hier in Weiß gehalten. So wäre es auch keine Abweichung , wenn der errechnete IE Wert im Bolusrechner weiß sein wäre.

 

Ein weiterer Punkt, der mir noch auffiel, ist lediglich der Curser in den anzutippenden Feldern z.B. wenn ich im Bolusrechner meine KH eintragen möchte, erscheint eine zarte Umrandung des Feldes und der Curser. Beide scheinen in bleu zu sein. Die Umrandung ist nicht wichtig, der Curser schon, damit ich weiß, dass ich im richtigen Feld bin. Auch hier wäre es schön, wenn dieser weiß und fett sein könnte.

Dann wäre ich sehr happy.

 

Ich hoffe verständlich geschrieben zu haben und bedanke mich sehr, dass das Thema hier bearbeitet wird.

Danke.

Alamo04, 2022-07-04:
@AnagarsWelt 
I understand your point, but how @Andries-Smit says, the colors are choosen to be fine for most of all users.

@osodebailar can perhaps do that changes, that he have in mind, but I don't believe that they change the whole color.

I think you must change that on your own in code.

I send you PM in Looper Community,  perhaps I can help you with that.

closed 2022-07-10 by MilosKozak