nightscout/AndroidAPS #1880: PasswordReset not working on AccuCheck Insight

opal06, 2022-07-03:
I tried to reset the master password on version 3.0.0.1 (Build: 479bfc4-2022.02.04-14:10) following the instructions given in the docs for v3.0 and above. I created the "PasswordReset" file in AAPS/extra and restarted the app. The file disappeared and the app showed a toast message stating that the password had been set. But the password still isn't the Insight's serial number. Looking at the logs, it seems the pump sometimes reports serial number as "null", so I tried that, but it didn't work either. 

I there something I'm doing wrong or overlooking? Or is there a different password for the Insight? 
Help would be really appreciated.

Philoul, 2022-07-10:
I checked master password reset on my dev phone (3.0.0.2-dev-n version) and it worked fine.
Are your access to preferences locked or not ?
If unlocked can you select Insight Driver > Preferences > Insight pairing
=> You will see on the first line the serial number recorded (should be the same serial number writen in the back of your pump...)

opal06, 2022-07-11:
Interesting... Yes, I can access settings and I have used the exact string you advised. I might build the dev version and check out if it works with it.

Philoul, 2022-07-11:
@opal06 The actions done during my verification was:
1) connect phone to my computer to copy an empty file PasswordReset in extra folder
2) exit aaps from 3 dots menu
3) launch again aaps
4) open 3 dots menu and select Preferences > General > Master password
5) Aaps requests a password on a popup to change it : I entered pump serial number and Ok
6) then I got the 2 fields to be able to enter a new Master password.
7) After validation I could recover full access to aaps with my new Master password...
Of course, it's important to export your settings immediatly after with your new password (all your old exports cannot be imported without your old password)...

Can you check if it works on your side using the same process?

closed 2022-07-15 by MilosKozak