nightscout/AndroidAPS #1885: 3.0.0.2 dev-n TDD not agree with pump

chrisbev, 2022-07-04:
3.0.0.2 dev-n TDD on Action Tab
In 2 days of the 7 the TDD in AAPS did not agree with the  pump. 
One day had a red line the other day had exactly the same data and both about half that in the pump pump?  Pump on 1st =25.1U & 30th=19.6U. See attached picture.  Insight pump.  Phone updated to Android 12 noon on 2nd. History/reason: Running dISF last night I was constantly  below target and dISF sensitivity 1.22 so AAPS had reduced target, increased basal and ever growing negative IOB.  Looking at the tab I suspect this was due to my slightly higher TDD the day before and a lower weekly TDD due to the wrong entries.  NB  I do seem to get more AAPS restarts with dev-n assuming the pump TDD is read near midnight then doesn't seem to correlate.
Samsung A41, xDrip, Patched LibreLink.

[image]

[image]
[image]
[image]

[AndroidAPS._2022-06-29_10-01-29_.3.zip](https://github.com/nightscout/AndroidAPS/files/9041490/AndroidAPS._2022-06-29_10-01-29_.3.zip)
[AndroidAPS._2022-06-30_00-49-05_.0.zip](https://github.com/nightscout/AndroidAPS/files/9041491/AndroidAPS._2022-06-30_00-49-05_.0.zip)
[AndroidAPS._2022-06-30_00-49-05_.1.zip](https://github.com/nightscout/AndroidAPS/files/9041492/AndroidAPS._2022-06-30_00-49-05_.1.zip)
[AndroidAPS._2022-06-30_10-02-22_.4.zip](https://github.com/nightscout/AndroidAPS/files/9041493/AndroidAPS._2022-06-30_10-02-22_.4.zip)
[AndroidAPS._2022-07-01_00-00-04_.0.zip](https://github.com/nightscout/AndroidAPS/files/9041494/AndroidAPS._2022-07-01_00-00-04_.0.zip)
[AndroidAPS._2022-07-01_10-02-24_.0.zip](https://github.com/nightscout/AndroidAPS/files/9041495/AndroidAPS._2022-07-01_10-02-24_.0.zip)

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-07-15:
can you verify bolus amounts on treatments tab for these 2 days?

chrisbev, 2022-07-15:
Quick tot up I make it 30th=10.37U and 1st=15.00U, screenshots and part csv file attached:-
[image]
[image]
[2022-07-15_094238_UserEntryPart.csv](https://github.com/nightscout/AndroidAPS/files/9119854/2022-07-15_094238_UserEntryPart.csv)

Philoul, 2022-07-18:
Got the same issue 
Data is ok for almost all days except:
- 12/07: half the amount of my pump
- 11/07: in red same values than 12/07
- 10/07: day missing in TDD
[image]

Philoul, 2022-07-18:
I never use this TDD button because results are less accurate than values in stats.
- Priming done from aaps are included in TDD and bolus in pump (so every 3 days inconsistant values)
- extended bolus (fake TBR) are in bolus in pump and in basal in stats.

chrisbev, 2022-09-24:
3.1.9.3-dev-a (with dISF_2.0) another occurance. Can't quite see how could affect it as pump data OK but on 22nd failed to upload data to Glooko (as required by hospital) not even after loads of tries (can take several attempts for me) but did succeed after a few tries on 23rd? The handset is connected to pc for upload.  As handset needs bt to pump I do have them both paired in pump and only turn on remote for upload. Handset appears to get data OK presume when AAPS/phone not communicating?
[image]

Philoul, 2022-10-03:
Unfortunatly I don't know how to fix it...
I got this issue only once 3 months ago...

I can try a code analysis to understand how TDD button works 🤯...

But I don't understand what's the added value of this button (compare to Statistic Activity) 🤔

chrisbev, 2022-10-04:
Hi so is the Actions tab TDD filled by the pump or from AAPS summing insulin delivery?  Perversely I just had my database reset and only lines after the reset are there but I still have the data in my pump or does the pump driver only add new TDDs as well each day/night?  Any thoughts what it means if get a red line?