nightscout/AndroidAPS #1887: TBR set for 120 seconds with Orangelink instead of 120 minutes

Muppet1313, 2022-07-05:
Reporting bugs
--------------

Pump sets 120 minutes Temporary Basal rate 0.000 U for 120 minutes-> 
08:14:52.614 [Thread-15922] I/PUMPCOMM: [MedtronicCommunicationManager.setTemporaryBasal():611]: setTBR: Rate: 0.000 U, Duration: 120 min
08:14:54.125 [Thread-15922] I/PUMP: [MedtronicPumpPlugin.setTempBasalAbsolute():761]: MedtronicPumpPlugin::setTempBasalAbsolute - setTBR. Response: true

NS says 120 seconds ->

08:14:54.134 [Thread-15922] D/CORE: [PumpSyncStorage.addTemporaryBasalRateWithTempId():179]: PumpDbEntryTBR: PumpDbEntryTBR(temporaryId=20220705081454, date=1657001694125, pumpType=MEDTRONIC_554_754_VEO, serialNumber=856911, pumpId=null, rate=0.0, isAbsolute=true, durationInSeconds=120, tbrType=NORMAL)

[image]
[AndroidAPS_LOG_1657001820157.log.zip](https://github.com/nightscout/AndroidAPS/files/9045769/AndroidAPS_LOG_1657001820157.log.zip)
[image]

andyrozman, 2022-08-20:
Are you sure this doesn't work? I haven't made any changes in the code there.... So when this is set:
1. What is set on pump?
2. What is shown in Treatments (in AAPS)
3. And what is shown in NS...

Muppet1313, 2022-08-20:
unfortunately I can't reproduce it now the error message because we have switched to the Dana I pump. I will close the ticket.

closed 2022-08-20 by Muppet1313