nightscout/AndroidAPS #1891: AAPS 3.0.0.2 Dash Pod Activation problem

reach07steve, 2022-07-06:
Problem occurred: July 5, 1400 PST

I have a Samsung A21 phone, Dexcom G6, using BYODA which feeds NS and XDrip. I activated my first Dash pod about a week ago, no problem. I was unable to activate the second pod 3 days ago. Today (July 5, 1400 PST) I tried a third pod with no success.  Today I successfully deactivated the used pod and put it far away from me (in another building).  I powered off my Omni pdm. I retried activation using AAPS on my phone about 10 times (pressing Retry button), until I got this error (screenshot above):  "Generic error: java.lang.illegalStateException: Trying to set Bluetooth Address to E2:D9:61:24:51:E0, but it is already set to CA:C8:61:24:51:E0" .  I have several  Bluetooth devices paired to my phone, but as far as I know none were on when I was trying to activate. 

Would much appreciate any help someone can offer. Apologies I can't figure out how to paste my log file here.

Reporting bugs
--------------
[image]

- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

vanelsberg, 2022-07-06:
Q: What Android version is your phone running?

vanelsberg, 2022-07-06:
On some phones permissions are reset after not using an app for some time.
Check permissions on the AndroidAPS app:
[image]

reach07steve, 2022-07-06:
Android Version 12

 

Regards, Steve

 

CONFIDENTIALITY NOTICE

This email may be privileged and confidential.  Any use or dissemination of this information by a person other than the intended recipient is not authorized.  If you have received this message in error, please notify the sender immediately by reply email and delete the message from your system.  Thank you.

reach07steve, 2022-07-06:
Many thanks for your help. Attached is the log file from July 5. I tried to make the activation around 1400.

Regarding your other questions: 
Android version 12.
When I got the activation error I had put the deactivated pod about 20 feet away inside my microwave.
Checked AAPS permissions and all looks good. 
Since Android 12 it is harder to find the system apps to clear the BT cache. Still trying to figure that out. 

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/9056168/AndroidAPS.log)

reach07steve, 2022-07-06:
> Could you please add logfile from the time this problem occurred?
>
> The error message suggests AAPS is seeing an unknown DASH Pod in the neighborhood of your phone (using the BT address CA:C8:61:24:51:E0). Could that be?
>
> Make absolutely sure there is no DASH Pod around that is activated (or waiting for activation)
> Try resetting your BT system app (settings->app (show system apps) -> clear cache+memory, force close - then reboot the phone).

Theodoor, here is the log file. I tried the activation at about 1400 hrs. 

 

Regards, Steve

 

CONFIDENTIALITY NOTICE

This email may be privileged and confidential.  Any use or dissemination of this information by a person other than the intended recipient is not authorized.  If you have received this message in error, please notify the sender immediately by reply email and delete the message from your system.  Thank you.

vanelsberg, 2022-07-08:
According to the logs AAPS could not "see" the new Pod when trying to activate.
Do try resetting the Bluetooth app and reboot the phone.

What region are you in?
We found Pods running older Firmware versions are still distributed in CA and the US. We have reports from users running Android version 11 and/or 12 experiencing the problem where specific phones are unable to "see" the DASH Pod.

reach07steve, 2022-07-08:
Thanks again Theodoor. Your message is timely, I am just about ready to change my pod. I am in BC, Canada.  I am on a Samsung A51, Android 12. Rebooting the phone, not a problem. Not sure what you mean by “resetting the Bluetooth app”.

 

Regards, Steve

 

CONFIDENTIALITY NOTICE

This email may be privileged and confidential.  Any use or dissemination of this information by a person other than the intended recipient is not authorized.  If you have received this message in error, please notify the sender immediately by reply email and delete the message from your system.  Thank you.

reach07steve, 2022-07-13:
Theodoor, on my last pod change, July 11 I had the same problem with the same error. I followed all of your instructions (see below)  except I was not able to clear the Bluetooth cache. They have made finding system apps more difficult in Android 12, and I can't figure out how to get there.   I am assuming that I am experiencing what you suggested earlier:  "We found Pods running older Firmware versions are still distributed in CA and the US. We have reports from users running Android version 11 and/or 12 experiencing the problem where specific phones are unable to "see" the DASH Pod."  

Any suggestions on how to resolve this ?  I am considering buying a new phone but I am concerned that the combo of old Firmware and Android 12 may still cause me problems.

Activation process, July 11:

1 Deactivated old pod with pdm (because it was started with pdm)
2 Went to another building 40 feet away from the deactivated pod and pdm. 
3 Ensured there were no active Bluetooth devices or pods (old or new) in this building, except for my sensor.
4 Rebooted phone (Samsung A21)
5 Started activation using AAPS
6 Filled pod
7 AAPS could not locate a pod to activate
8 Pressed Retry about 10 times until I got the same error as in my previous, failed activation (see below)
9 Rebooted phone again and retried activation in AAPS. Retried about 10 times until I got the same error: "Generic error: java.lang.illegalStateException: Trying to set Bluetooth Address to E2:D9:61:24:51:E0, but it is already set to CA:C8:61:24:51:E0"

vanelsberg, 2022-07-13:
> ... I was not able to clear the Bluetooth cache. They have made finding system apps more difficult in Android 12, and I can't figure out how to get there. 

System->apps: Look for something like this:?
[image]
[image]

reach07steve, 2022-07-13:
Thanks again. I am making some progress but apologies if I am being thick headed here. When I Show system apps I see 3 BT items as shown in the screen shot. None of them has an option to "clear cache". I can Force Stop on any of them. I must be missing something. Please let me know what I should be looking for. 

[image]

vanelsberg, 2022-07-13:
Github is not the proper place for discussing this, please see my reply/addtional comments on the Discord channel?

vanelsberg, 2022-07-13:
> I am assuming that I am experiencing what you suggested earlier: "We found Pods running older Firmware versions are still distributed in CA and the US. 

Looks like it (unless restting the BT app solves...)
Unfortunately this is Pod/Phone hardware related and we have no solution for this problem. Sorry...

You could other phone hardware (brand/type) to see if it does recognise your Pods.

closed 2022-12-15 by MilosKozak