nightscout/AndroidAPS #1899: Lineage OS, ANDROID 10 AAPS 3.1 DOESN'T WORK

agniabakush, 2022-07-08:
Reporting bugs
-----------
Application closes after launch with the message "An error has occurred in the application"

Andries-Smit, 2022-07-08:
Can you add log details?
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html

agniabakush, 2022-07-11:
[AndroidAPS._2022-07-07_17-27-07_.0.zip](https://github.com/nightscout/AndroidAPS/files/9080405/AndroidAPS._2022-07-07_17-27-07_.0.zip)
[AndroidAPS._2022-07-08_09-29-56_.0.zip](https://github.com/nightscout/AndroidAPS/files/9080406/AndroidAPS._2022-07-08_09-29-56_.0.zip)

agniabakush, 2022-07-11:
I don't know exactly which one, but hope these help.

MilosKozak, 2022-07-11:
hmm I see you are not using master (or latest dev) version.
This version is not supported here

closed 2022-07-11 by MilosKozak