nightscout/AndroidAPS #1901: New crashes

elgrupo123, 2022-07-09:
We've been getting new reboots a lot lately. 
This morning at 7:43 am. 
Maybe you can see from the log files what the problem is.

Thanks

ADD TIME OF EVENT HERE:
ADD ISSUE DESCRIPTION OR GITHUB ISSUE REFERENCE NUMBER:
-------------------------------------------------------
(Please remember this will send only very recent logs.
If you want to provide logs for event older than a few hours,
you have to do it manually)
-------------------------------------------------------
AndroidAPS 3.0.0.2-dev-n
Build: ad89d5431-2022.07.02-07:56
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout-Version: 14.2.5
Entwickler-Modus aktiviert
[AndroidAPS_LOG_1657349580752.log.zip](https://github.com/nightscout/AndroidAPS/files/9076078/AndroidAPS_LOG_1657349580752.log.zip)

agmsd, 2022-07-09:
The same here. I think there cause is June update.

agmsd, 2022-07-10:
> The same here. I think there cause is June update.
I fixed the problem by flashing a factory image. It's running smoothly now.

MilosKozak, 2022-07-10:
sorry nothing is visible from log file
PS: turn OFF autosens logging

closed 2022-07-10 by MilosKozak