nightscout/AndroidAPS #1904: The complication doesn't open AAPS on Wear OS 3

agmsd, 2022-07-11:
The complication doesn't open AAPS on Galaxy watch 4. It takes a lot of clicks to open.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

Andries-Smit, 2022-07-11:
@agmsd can you set change the action to another preference and back?
*AAPS > Settings >Complications Tab Action*. Select **MENU**, and then back to **DEFAULT**
Does your issue still occur after changing the tab action?

MilosKozak, 2022-07-13:
the principle didn't change. was is working before?

Andries-Smit, 2022-07-13:
It works Before Android 10 (API level 29) 
With the new restrictions, it does no longer work on newer devices like the watch 4.
Though I did not test with older Aaps versions

mk89pwnz, 2022-10-14:
I have same problem but its more then that, personally if I use default Samsung watch ace with compilations AAAP just stops working. It shows all information like BG, IOB, COB but when I try to set TT or do a bolus request it's sent but I never get the confirmation. Only way to get it workings is to use default AAPS watchfaces. If i forcestop AAPS from the phone and start it again it will work for few min or hours and then stop again, I beleve this is samsung bug but maybe it could be fixed by AAPS?

agmsd, 2022-12-07:
> the principle didn't change. was is working before?

I bought my Galaxy Watch 4 Classic and I started using it with version 3.0, so I'm not sure! bolusing works sometimes disconned and reconnect the Bluetooth for once or never!. The only thing working very well now is the BG readings.

agmsd, 2022-12-07:
I can doing the testing whenever you need me to!

agmsd, 2023-04-11:
I think the problem is worth the Bluetooth on pixel 6 and 7. When I reset the Bluetooth, watch AAPS runs once or twice then stops running after that.

agmsd, 2023-05-04:
Finally, it's working now with May wear os update!

MilosKozak, 2023-08-17:
Closed for inactivity. Create new issue is if you still see the problem on latest branch

closed 2023-08-17 by MilosKozak