nightscout/AndroidAPS #1906: Redefine the Objective question: Insulin delivery/entry errors

koelewij, 2022-07-12:
Reason for this proposal is that the question is very vague and depending on the result of one answer, 2 other answers might apply.
Almost everyone is struggling with this question, and hints can only be given correctly if you know the plannend answers. 
Question:

- "What should you do if you received less insulin than the pump history suggests eg. due to an occlusion, a failed canula or forgetting to reattach the pump after a shower?"

When you take a look at the question i see the following issues:

1. It assumes you know this is the case. 
2. One of the required answers is to "Compare values in AndroidAPS and pump history (if pump supports this)", while the question states: that you got less insulin than the pump history suggests, so the answer should be FALSE instead of true.
3. The second answer: "Bolus a proportion of your calculated 'missed' insulin by either seringe/pen or using a prime." can only be the TRUE if you would be aware of the amount of missed insuline, which you could NOT determine, since your pump history is not showing the correct value. 
4. If the missed insulin is marginal, you could consider the answer where you do nothing and AAPS takes care of it. 

Therefore it would be my proposal, to rephrase it into something like:

- "What should you do if you received less insulin than the pump history suggests eg. due to an occlusion, a failed canula or forgetting to reattach the pump after a shower, and you were able to determine the amount of insulin you are missing is substantial and needs action?"

With the following answers:

1. Bolus a proportion of your calculated "missed" insulin by either seringe/pen or using a prime.
2. Do nothing - AndroidAPS will make the appropriate adjustments.
3. Delete Insulin data from Nightscout Careportal to remove it from the pump history.
5. Bolus your calculated "missed" insulin through AndroidAPS to compensate for the missing insulin.

I presume it is clear, what the correct answer is.  

I am open for discussion on this topic.

MilosKozak, 2023-02-14:
goal of this question is to verify real situation when any error occur. You can create PR with now proposal

koelewij, 2023-02-14:
In my work a PR is a Problem Report, but I presume you are referring here to a Pull Request. My knowledge of a pull request is zero, but I expect that as a developers you revise code and publish this in a kind of pull request. Modifying code and releasing that here is something I am not able to do. I thought it was possible to mention an enhancement here and that depending on priority, this would flow in to new code. Now I get the feeling I have to change the code myself.