nightscout/AndroidAPS #1915: WIDGET / Android 12 : No more parameters

robertrub, 2022-07-16:
I changed phones for a POCO F3 on Android 12 and I can't get the widget parameters any more. Long press on widget gives me only the size points but no parameter menu.

[image]

MilosKozak, 2022-07-16:
probably too small screen. make the widget smaller first

closed 2022-07-16 by MilosKozak

robertrub, 2022-07-16:
Nice try @MilosKozak  but no,. I had already tried to reduce widget size and the screen is big and has a huge definition. Please reopen the issue ;)

[image]

[image]

MilosKozak, 2022-07-16:
[image]
it works well even on Android12

robertrub, 2022-07-16:
Probably it is due to the _MIUI 13_ overlayer on Android 12 of my XIAOMI POCO phone then :(