nightscout/AndroidAPS #1919: Autotune is no longer selectable?

vanelsberg, 2022-07-19:
**AndroidAPS 3.1.0**
**Build**: bc2cd2c59
**Flavor**: Fullrelease, developer mode

**Problem description:**

Autotune is somehow "disabled"?
On upgrade, tab was removed. Option is still in config builder but is "unselecting" on selection

vanelsberg, 2022-07-19:
@MilosKozak Reproduction should be easy.
If not, let me know. I'll add logfiles

Philoul, 2022-07-19:
not on a released version (currently dev  build is 3.1.0 so master release ref).
Autotune is only available on dev with Engineering mode.

vanelsberg, 2022-07-19:
@Philoul 
>  Autotune is only available on dev with Engineering mode.

Dev branch was just bumped to version 3.1.0 so yes, this is about latest dev commit.

osodebailar, 2022-07-19:
@what is the reason to not make it available to the next master ?
Plug-ins like Autotune and the new Dynamic ISF are working perfect and very stable .

0xAalaoui, 2022-07-19:
Fixed by replacing this line https://github.com/nightscout/AndroidAPS/blob/dev/app/build.gradle#L108 with 

> version "3.1.0-dev"

MilosKozak, 2022-07-19:
it's already changed to dev again

closed 2022-07-19 by MilosKozak