nightscout/AndroidAPS #1921: 3.1 crashes instantly on Pixel 6

dottedfingertips, 2022-07-19:
Reporting bugs
--------------
- I built an apk based on 3.1 (master).  When I try to open it on my Pixel 6 it crashes instantly.  I reboot the phone several times but had the same issue.    I also tried uninstalling AAPS and then installing 3.1 but had the same result.
- Build Version - 3.1
- I mentioned this on discord and was asked to open this issue without logs as this may be a general issue with this phone.

dottedfingertips, 2022-07-19:
Some additonal info on the phone build:

Android Version: 12
Security Update: 5 June 2022
Google Play system update: 1 June 2022
Baseband version: g5123b-100840-220505-B-8544885
Kernel Version: 5.10.81-android12-9-00071-g64099431c232-ab8394429
Build number: SQ3A.220605.009.B1

MilosKozak, 2022-07-19:
do you use white theme?

dottedfingertips, 2022-07-19:
> do you use white theme?

Yes, I've been using the white theme in the dev version.

MilosKozak, 2022-07-19:
seems to be issue on some phones. switch to dark for now

dottedfingertips, 2022-07-19:
> seems to be issue on some phones. switch to dark for now

Thanks for that.  I'm now up and running on 3.1.  For anyone else having this problem here are the steps I took:

*  Build and installed the apk using the 3.1 beta 1 branch.  This was the only version I could find that would open on the phone without crashing (both dev and master crashed straight away).
*  Turned on dark mode in Preferences.
*  Built and installed 3.1 (master).  This now launches and works as expected.  Verified with a test bolus that worked with my Medtronic pump.

I'm happy to test dev when a fix for light mode is ready.  I did really like that in fairness.

drewholmes, 2022-07-20:
I switched to device specific on the P6 Pro and instantly having the same issue. Quick switch over to dark mode on the phone let me back into the app and then switched back to dark mode always in the app.

closed 2022-07-20 by MilosKozak