nightscout/AndroidAPS #1922: v3.1 can't select DynISF

robertrub, 2022-07-20:
I can't select DynISF in v3.1. When I hit it, AAPS selects the OpenSMB AMA option.

[image]

https://user-images.githubusercontent.com/76626474/179896862-d89cdd67-f5e6-48c4-b28d-645b6d08b341.mp4

koelewij, 2022-07-20:
> I can't select DynISF in v3.1. When I hit it, AAPS selects the OpenSMB AMA option.
> 
> [image]
> 
>  Screenrecorder-2022-07-20-06-23-18-679.mp4

Having the same issue here.

MilosKozak, 2022-07-20:
use dev branch. this is not included in master version

closed 2022-07-20 by MilosKozak