nightscout/AndroidAPS #1933: Unable to sync AAPS 3.1 with Sony Smartwatch 3

JimZam, 2022-07-20:
Good night. My Sony Smartwatch 3 doesn't get data from AAPS once updated (the phone and the watch) to version 3.1. I have tried every single thing I can think of to make it work with no luck. If I downgrade to 3.0.2 (once again the phone and the watch) everything works fine. The app works in the watch and of course on the phone, but they don't sync. In AAPS it says "No watch connected" but I get notifications and data from xDrip+ and, as I said, if I downgrade it works again. I have read the logs but it's too much to me, I'm afraid I don't understand anything :-(
Any help is welcome.
Thank you in advance.

[image]
[image]

MilosKozak, 2022-07-21:
show screenshot of version installed in watch

JimZam, 2022-07-21:
Good morning. Here you are.
Thank you so much Milos.

[image]
[image]

MilosKozak, 2022-07-21:
watchface is 3.0.0.2

JimZam, 2022-07-21:
Sorry, I have been trying so many builds I didn't remember which one was installed...
Here you are
[image]
[image]
[image]
[image]

JimZam, 2022-07-21:
I'mafraid it is the same version, isn't it? Then version 3.1 gives me a black screen. I have two builds and probably one of them is 3.0.02, the one in the photos; the other one, 3.1, doesn't work at all :-(

JimZam, 2022-07-21:
No way to make 3.1 work on the watch, it crashes or gives a black screen.
Sorry for the inconvenience and thanks for your help.
Kinzd regards

closed 2022-07-21 by JimZam