nightscout/AndroidAPS #1938: Bugs in 3.1.0 to be fixed

yhonatanb20, 2022-07-21:
Please pay attention to these while working on fixed version:
1. App crushes when set to light theme
2. Bugs in basal displaying, displays 0.39 U/h instead of 0.40 U/h on main screen
3. This new blue slider is really annoying, just stops in the middle or blinks over and over again, pls remove it
4. Pls chage the TT/Profile banner color, that yellow has very high contrast with the black amoled background, and the TT/star signs on it are white which doesn't look good either. Maybe some blue color would be good but really annoying this way
I had to downgrade back to 3.0.0.1 to avoid those issues...
Thank you for all your work, please fix these issues so that the next version would be more reliable and aesthetic.

osodebailar, 2022-07-21:
1. Is fixed since yesterday 
3. What do you mean with "blue slider"? The  progress bar on the main page?
4. The high contrast is wanted. Maybe it's higher on amoled, on other displays it's look very well.

yhonatanb20, 2022-07-21:
2. Yes, exactly, the progress bar. Just stucks in the middle of progress or goes over and over again, pls at least make it optional.
3. By amoled I meant the new design. On 3.0.0.1 where the beackground is grey the yellow looks OK, on 3.1.0 where background is black and the yellow is more bright it doesn't look good togeather, some "bee design", pls at least make another softer color optional.

osodebailar, 2022-07-21:
Anyway I think it would be a good idea if you seperate the 3 issues, so that the developers can handle it seperate and decide what to do or what not, and if when.

closed 2022-07-21 by yhonatanb20