nightscout/AndroidAPS #1939: manual BG checks keep replicating in Treatments

justmara, 2022-07-21:
I'm using xDrip to add manual BG checks. xDrip then uploads it to NS. Both nsclient and aaps itself (checked in 3.0, 3.1-beta1, 3.1) keep syncing this value and re-adding it to Treatmens list.
Here is how it looks in NS database (you can see only 1 value)
[image]
and here how it looks in nsclient's Treatments
[image]

I tried deleting all the duplicate entries trough treatments 'delete' button, but they keep re-appearing again after some period of time. After 'deleting' these entries are marked `IsValid: false` in mongodb, but still keep downloading and adding back to treatments list.

MilosKozak, 2022-10-04:
only one app must upload to NS, all other configurations usually lead to troubles you see
it's mentioned in wiki

closed 2022-10-04 by MilosKozak

justmara, 2022-10-04:
What do you mean under "only one app must upload to NS"? NS is and API. It receives commands and returns data for requests. There is no (and never was) any reason on limiting number of applications uploading data to NS. The only rule was to not duplicate data.
Here is the case, where data is NOT duplicated in NS. It IS duplicated only in AAPS itself. NS contains and shows single record.
But AAPS continuously syncs it to local db. Over and over again. Duplicating same record from NS to new records in LOCAL treatments db.
So its not even close to _completed_. Same issue is observed by other users, who has their xDrip sending BG and BG Check records directly to NS.

MilosKozak, 2022-10-04:
problem is data from all app are not 100% compatible, API v1 doesn't have sync protocol and finaly AAPS is using websockets and data comming through WS are even modified.
If there would be an easy solution it would already in place