nightscout/AndroidAPS #1941: PasswordReset Bug Dana-i Ver. 3.0.0.2

Chikibai, 2022-07-21:
Bug Info here: [https://github.com/nightscout/AndroidAPS/issues/1916](url)

...Password is empty after reset, not SN...

After Reset I set no new Pw (leave it empty). I make an Export of my Settings. Now i want to Import my Settings and get the info that my file is manipulated, no Import.
What can i do, to get my settings back.

I try do re-install the app no success.

MilosKozak, 2022-07-21:
sorry you cannot make export without password (it's used for encryption)

Chikibai, 2022-07-21:
Export with empty password, after Reset works. 

During Import i have to insert 2 times the empty Password. It ends with a popup windows.

MilosKozak, 2022-07-21:
I tested current version 3.1 now and export with empty password is not possible

Chikibai, 2022-07-21:
Export with empty password works in 3.0.0.2. You fixed this bug in 3.1 that sounds good. 

Sorry, but i need an import with empty password.

My export-file has about 520 lines of settings, do you need this?

Is there a rescue option?

MilosKozak, 2022-07-21:
sorry, no idea what you did and what to do now because master password is used for encryption

closed 2022-10-04 by MilosKozak