nightscout/AndroidAPS #1947: max_iob = 0 after upgrading to AAPS 3.1.0? Maybe due to new objectives questions?

dhermanns, 2022-07-22:
Looks like there are objectives questions. After upgrading to 3.1.0 they show up:

[image]

And I noticed that the restriction to max_iob = 0 seems to apply:
[image]
[image]

I already noticed that when upgrading to 3.0.0.2. Answered the questions correctly. But that didn't seem to remove the max_iob restriction. 

Now after upgrading to 3.1.0 there are the new questions.
But I fear that even after answering them correctly, the restriction will still persist.

Any thoughts?

MilosKozak, 2022-07-22:
what do you have in preferences?

dhermanns, 2022-07-22:
Hi Milos! In preferences a max IOB of 9 is configured.

agniabakush, 2022-07-23:
You should chech max iob value in AMA module. It seems APS apply min value even module is inactive

dhermanns, 2022-07-23:
Thanks - this is configured to 9 too:

[image]

Chikibai, 2022-07-23:
Autosens ist not checked on.

In objectives 3 (Ver. 3.1)there are new questions (not translated). I remember in Ver. 3.0.02 this questions are I:C Verhältnis...now splitted in Ver. 3.1

dhermanns, 2022-07-23:
Double checked the objectives and turns out the COB questions are once more marked as unsolved :-/

Did a export/import after reinstalling. But looks like that doesn't run smoothly.

Chikibai, 2022-07-23:
Are you on latest 3.1.0.2 Ver. 3.1.0 have bugs

dhermanns, 2022-07-23:
Missed the new version yesterday ;-) . Now I updated. But still the same behavior. All questions still answered. But max_iob seems still stay at zero. Tried to update it both in AMA and SMB settings. But no change.

Looks like its still disabled as if the objective 3 questions still unsolved.

dhermanns, 2022-07-24:
Looks like the LGS constraint seems to set my max_iob to 0:

[image]

Thinking about disabling this constraint completely to see whether this helps. 

Thinking of disabling it in the SafetyPlugin.kt:

```
    override fun applyMaxIOBConstraints(maxIob: Constraint<Double>): Constraint<Double> {
        val apsMode = sp.getString(R.string.key_aps_mode, "open")
        val maxIobPref: Double = if (openAPSSMBPlugin.isEnabled() || openAPSSMBDynamicISFPlugin.isEnabled()) sp.getDouble(R.string.key_openapssmb_max_iob, 3.0) else sp.getDouble(R.string
        .key_openapsma_max_iob, 1.5)
        maxIob.setIfSmaller(aapsLogger, maxIobPref, rh.gs(R.string.limitingiob, maxIobPref, rh.gs(R.string.maxvalueinpreferences)), this)
        if (openAPSAMAPlugin.isEnabled()) maxIob.setIfSmaller(aapsLogger, hardLimits.maxIobAMA(), rh.gs(R.string.limitingiob, hardLimits.maxIobAMA(), rh.gs(R.string.hardlimit)), this)
        if (openAPSSMBPlugin.isEnabled()) maxIob.setIfSmaller(aapsLogger, hardLimits.maxIobSMB(), rh.gs(R.string.limitingiob, hardLimits.maxIobSMB(), rh.gs(R.string.hardlimit)), this)
        if (openAPSSMBDynamicISFPlugin.isEnabled()) maxIob.setIfSmaller(aapsLogger, hardLimits.maxIobSMB(), rh.gs(R.string.limitingiob, hardLimits.maxIobSMB(), rh.gs(R.string.hardlimit)), this)
        if (apsMode == "lgs") maxIob.setIfSmaller(aapsLogger, HardLimits.MAX_IOB_LGS, rh.gs(R.string.limitingiob, HardLimits.MAX_IOB_LGS, rh.gs(R.string.lowglucosesuspend)), this)
        return maxIob
    }
```

Would this have any bad side-effects?

dhermanns, 2022-07-25:
I switched MAX_IOB_LGS to 9.0. Closed Loops seem to work as expected now.

One-million-dollar question is: Why thinks AAPS that the Loop should be in LGS Mode?
Looks like a bug to me...

Chikibai, 2022-07-25:
max_iob = 0  and LGS is part/target of objectives 6

dhermanns, 2022-07-25:
Hm - maybe I should restart and reanswer objectives 6 then?

Which code does the objectives 6 check? I could try to deactivate it there and try to drilldown the real cause...

Sabr1naT1, 2022-07-29:
Hi @dhermanns, 
I have the same issue with max_iob, could you please give an update whether restarting objective 6 fixed it?

dhermanns, 2022-07-29:
Hi! No, not yet. We need the closed loop badly right now. So it will take some time for me to reset objectives 6.

Did you had unanswered questions too before updating to a newer version?

Sabr1naT1, 2022-07-29:
Currently I'm on version 3.0.0.2. I did not update the app, just installed and did the objectives one after the other and I'm stuck at objective 7 right now, because max_iob is 0 no matter what i'm changing in the settings.
I want to update on 3.1 on sunday and if this will not fix objective 7, then I'm going to restart objective 6

dhermanns, 2022-07-29:
So you did a fresh answer of all questions using 3.0.0.2? Then my wild guess seems to be wrong and restarting objectives 6 won't help.

At least it would surprise me ;-). You could try make workaround. Works fine for me, but sure this is dangerous and comes with no warranty ;-)

Sabr1naT1, 2022-08-01:
Hi, after installing 3.1.0.2 today and importing my settings, I also had 3 unanswered questions in objective 3 (about COB, see screenshot). After answering them, I tried to finish objective 7. Additonal to "max IOB" appeared the point "Closed mode activated". And in the settings about Loop APS mode, I found the setting to change it from LGS to Closed Loop and - et voîla - objective 7 is green \o/
[image]
[image]

dhermanns, 2022-08-01:
Great - thanks. I checked your last setting and totally missed that one. Didn't remember that you could completely deactivate the closed loop here, too.

So the missing objectives haven't been the cause.
Case can be closed. Misconfiguration on my side ;-).

closed 2022-08-01 by dhermanns