nightscout/AndroidAPS #1948: Both 3.1.0.1 full-release and its dev have a bluetooth permission error

T1DLiberation, 2022-07-22:
I updated 3.1.0.1 to resolve 3.1.0's theme color triggered infinite app crashing. Theme color problem has resolved by 3.1.0.1. However, a bluetooth permission error has occurred. Both full-release 3.1.0.1 and 3.1.0.1-dev have a bluetooth permission error. Screenshot attached 3.1.0.1 full-release only though.
[image]

I tried finding out the settings but there is no exact option that I can enable bluetooth of AAPS as an end user. Currently, I have rolled back to 3.1.0-dev. 

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

heikeov, 2022-07-22:
Hi,
I have the same problem on an Unihertz Jelly 2 (Android 11) and Cubot KingKong2 (Android 10)...

MilosKozak, 2022-07-22:
what pump?

T1DLiberation, 2022-07-22:
Hi heikeov,
I think 3.1.0.1's code triggered this issue. If it is the case, it won't be a specific device' problem and it may occur in many various devices. I just downgraded to 3.1.0-dev version. If you stay on dark theme on 3.1.0.1, you may build and install 3.1.0 stable version from android studio that can enable the bluetooth as long as you can use it working properly one.

T1DLiberation, 2022-07-22:
Hi Kozac,
I use Dana R(English). Android 9.

heikeov, 2022-07-22:
I use Dana-i

heikeov, 2022-07-22:
I go back to 3101-dev_o.

heikeov, 2022-07-22:
I Ethik the problem occurs on Android 9 (TD1Liberation), 10 (me) and 11 (me).

T1DLiberation, 2022-07-22:
In my case, 3.1.0-dev built yesterday is fine.
However, both 3.1.0.1-dev and 3.1.0.1 stable version I built today have bluetooth permission issue.
3.1.0.1 has released just about 4 hours ago. If more people gets trouble, we will know what android versions they use.
So far, I use Android 9 and heikeov use 10 and 11.

MilosKozak, 2022-07-22:
ok, i can replicate on older phone.

T1DLiberation, 2022-07-22:
Acknowledged. Thank you.

MilosKozak, 2022-07-22:
fixed in master

closed 2022-07-22 by MilosKozak

T1DLiberation, 2022-07-22:
I have confirmed the issue has been fixed.
Please watch the build number on the 'About' section.
Thanks again for your dedicated commitment, Kozak.