nightscout/AndroidAPS #1952: request: bolus splitting

justmara, 2022-07-23:
There is a problem with large boluses for kids, or some insulins (fiasp for example): large doses hurts. This is mainly an issue for accu-chek because of its fast bolus delivery.
Tjis behavior forces to split large pre-meal boluses onto smaller parts: ie 0.7..0.9. But alllso here the duplicate bolus protection kicks in: you cannot make two similar boluses in a row. So you need (for example) to split 6u bolus into 0.7+0.8+0.7+0.8+0.7+0.8+0.7+0.8
This hurts (not phisically though :)).

Mybe it is possible to add a 'Split bolus' option to calculator, so it could automatically split total required bolus with respect to MaxBolus size and (-0.1) step to deal with repeating boluses protection?
Queue up these smaller boluses with shared progress bar and cancellation button for example?

Chikibai, 2022-07-23:
Do you use [Super Micro Bolus (SMB)](https://androidaps.readthedocs.io/en/latest/Usage/Open-APS-features.html)?

justmara, 2022-07-23:
Surely yes. But here I'm talking about meal pre-bolus.

Chikibai, 2022-07-23:
...Or you give, if necessary with pre-bolusing, a start bolus, which only partly covers the carbohydrates (e.g. 2/3 of the estimated amount) and let SMB fill up the rest...

does this help?

justmara, 2022-07-23:
I'm giving 100% pre-bolus, then lately write up fat/protein 'carbs' that I let SMB to deal with. 
Every UAM-based scheme with or without announcements fails on my kid because of Fiasp being too slow and my kid's absorption being too fast :) So only full pre-bolus can deal with food consumptions. We even make 5-10min pauses between bolus and food on evenings.
So when dealing with massive boluses - this problem occurs.
This is how our usual pre-meal bolus of 5u looks like
[image]
1u/1.1u is already sensitive but not so painful dose. Would be better splitting by 0.7/0.8, but kid is too hungry and splitting 5u this way takes ages :) (with Accu-Chek's tend to disconnects and long waits..)..

JimZam, 2022-07-24:
@justmara you can reduce the speed ofthe insulin injection in ypur Insight in Settings - Therspy settings - Bilus configuration - Delivery speed. In my experience, reducing the speed makes the bolus less painful.

justmara, 2022-07-24:
@JimZam we're on Accu-Chek Combo, not Insight. Combo doesn't have such option :(

Der-Schubi, 2022-08-01:
I have the same problem. Also using Combo but with Lyumjev. I would love such a feature!

justinthomas, 2022-08-11:
By the way, looks like this request could be a duplicate of https://github.com/nightscout/AndroidAPS/issues/1683.

justmara, 2022-08-11:
Ouch, didn't find that older discussion. But I see there @MilosKozak misinterpreted it with extended bolus. But these two are not the same. Extended bolus depends of pump's realisation - it is split for (generally) unknown amount of parts with unknown intervals.
But we're talking about common boluses: single big bolus split apart and delivered as chain of smaller ones. They all can be count as usual single boluses. There is no need for any aggregation or re-calculation. Only queue of smaller boluses. Maybe with some +-0.1 variation to evade `duplicate bolus` protection.

MilosKozak, 2022-08-11:
scheduling a bolus is a problem because you can never guarantee it will be performed. so many things can happen. to make it 100% reliable it would need pump support. but at the moment there is not such a pump as I know

justmara, 2022-08-11:
Noone is talking about 100% guaranteed delivery. even when you make single bolus you can get pump error and bolus will be canceled. It doesn't mean that boluses from app are useless :)
There is no need to maintain bolus queue upon app restarts. Maybe simple notification 'your queue wasn't completed' can be nice, but not mandatory.
All I want is avoid spending 10..15 minutes staring at phone screen and repeatedly counting boluses in my mind, entering new ones.

justinthomas, 2022-09-23:
I've been pondering ways to do this without requiring pump support. I'm hoping to someday start working on this, but I'm a C++ developer and new to this project, so it might be slow progress. In the meantime, I'd appreciate any feedback on the following approaches, especially in the context of the existing framework. And if anyone can suggest a starting point, that'd be super helpful.

Some ideas:

- Set a target delivery amount, suspend loop (until achieved or timeout), and deliver at the maximum SMB amount and cadence until the cumulative amount delivered reaches the target.
  - Could this be done using the Automation framework?
  - I think this method is the easiest to cancel (compared to following ideas) since "Cancelling" would just reset the target and resume loop (or whatever status was set prior to starting delivery).
- Schedule SMBs to achieve the desired bolus (not sure if SMBs can be scheduled or of the scheduling capabilities in Kotlin / Android). This may also be more difficult as the future SMBs would need to be cancellable, so we'd need some infrastructure to enable that.
- Set a high temp basal to achieve the desired bolus. Probably would need to suspend loop for that duration if the user isn't using SMBs.
  - I expect this would be limited by the pump's minimum duration for a temp basal. For example, Dash's minimum basal duration is 30 minutes, so a 7 U bolus would have to be delivered as a 14 U / hr temp basal over 30 min.

justmara, 2022-09-23:
>Set a high temp basal to achieve the desired bolus

This must be introducing some kind of protective time-window limitation maybe, and also will have to deal with current AAPS smb rate limitation (once in 3 minutes default, customizable via settings).

>Set a high temp basal to achieve the desired bolus

On the other side accu-chek combo, for example, has pump-level maximum level of TBS based on basal %. So you simply cannot set TBS high enough to deliver.
For example my kid's daily basal is 0.2, so maximum TBS is 500% (this also needs to be programmed to pump via IR cable before looping because default limit is 250% afair) which is 1u/h. This is not enough for sure.

MilosKozak, 2022-09-23:
i already answered. this cannot be done safe way

justmara, 2022-09-23:
by the way, [this](https://github.com/nightscout/AndroidAPS/blob/master/app/src/main/java/info/nightscout/androidaps/queue/commands/CommandSMBBolus.kt#L30) check can easily allow to queue up SMBs with timeout. for example `deliver up to total IOB of 6u in 5 minutes` with callback chain that checks current iob and sets DetailedBolusInfo. deliverAtTheLatest according original time of queue start. thats it. all the late boluses will be cancelled
sure there can be intersection of manual boluses and SMB from loop, but we can, for example, set temp target preventing SMBs for the period of delivering, or something like this