nightscout/AndroidAPS #1954: Sometimes AAPS loose synchronisation with BYODA and  remains 20/30min late

Philoul, 2022-07-23:
I started to have this issue with room update (but I don't know how to track or catch information to fix it...
=> BYODA always shows latest BG reading received less than 5min ago

Sometimes (not always), when I use my phone (web browser, phone call...) I can loose the synchro, and aaps is 10min, 15min, but it can sometimes be 50min or more late on BG value shown in overview, and it can be worse for latest loop run.
When I'm in this situation, I need to close all apps, leave AAPS in foreground and wait (after a while could be 15min, 30min, sometimes more), it resynchronize step by step (15min late, then 10 min late, then 5 min late, and Ok).
=> I never got this issue on v2.8.2 and ealier versions, it started with room update (when I worked on dev, then 3.0 and now 3.1) I still have the same BYODA version not updated recently (so nothing changed on BYODA side)

In screenshot below (web navigation and a phone call. I'm 20min late)
[image]

Log files (I don't know if it could help and what to select in maintenance to have correct information to track this)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/9173770/AndroidAPS.log)
[AndroidAPS._2022-07-23_00-00-05_.40.zip](https://github.com/nightscout/AndroidAPS/files/9173769/AndroidAPS._2022-07-23_00-00-05_.40.zip)

Philoul, 2022-07-23:
Same screenshot 35min later
[image]

[AndroidAPS._2022-07-23_00-00-05_.41.zip](https://github.com/nightscout/AndroidAPS/files/9173822/AndroidAPS._2022-07-23_00-00-05_.41.zip)

MilosKozak, 2022-10-04:
still valid with latest dev?

Philoul, 2022-10-05:
I updated recently my loop phone with latest dev.
I'll check again, when I loose byoda sync, how long it takes to synchronize again.

Philoul, 2022-10-06:
Unfortunatly yes...

Today my Phone was out during about 5h30 (battery empty).

Phone, AAPS and BYODA restarted and 21h10.
Less than 10 min after, I received in BYODA the history of missing data up to latest BG value (then a new value each 5 min in BYODA, OK)
AAPS received only  at 21h32 the history of data (with BG values up to "16 min ago") so 16 min late compared to BYODA latest received value.
It's only at 21h58 that I got the first run of the loop (so 48 min after the restart of AAPS and about 40 min after the resynchronization of BYODA with G6 sensor)...

[image]

gitthiy, 2023-03-04:
I can confirm this behavior [image]

In my case aaps is mostly 5 to 10 behind of actual bg measurement. 

I tried several things like
- reboot phone 
- stopping xdrip 
- stopping aaps

Sometimes aaps and xdrip are in sync, but after some time aaps is behind. 
Having issues that smb's are not really matching actual bg situation

closed 2023-08-17 by MilosKozak