nightscout/AndroidAPS #1956: [3.1.0] [master or dev] Error "Wrong pump device" on virtual pump after restart

rICTx-T1D, 2022-07-23:
* choose a virtual pump, subtype is unspecified 
* give some insulin or use calculator 
* quit the app by app-menu or android control panel or restart your phone 
* start aaps
* wait some times
* give some insulin or use calculator 
* error is present- Wrong pump device
* switch in configuration menu to a other physical pump and back to virtual pump 
* give insulin or use calculator 
* everything worked fine, until next app start

rICTx-T1D, 2022-07-23:
[image]

libxmike, 2022-08-01:
can report same issue on 3.1.02 (dev)

mpessnegger, 2022-08-02:
can report the very same behaviour on 3.1.0.2 (fullrelease). i just press "dismiss" and the app works properly (Accu Check Combo, Decom G6

rICTx-T1D, 2022-08-02:
Additional I will report, that when error message is shown,  the insulin injection was not save in database or nightscout. After switching pump driver to an other  and back, all working fine with saving the data.

muthijammy, 2022-08-07:
Every restart of AAPS 3.1.0.2 reports the same error. Changing to virtual pump & then physical pump fixes till next restart. Not sure if it is impacting any other functionality. But error msg fills the screen and so graph1 on home screen is truncated on s9 plus mobile which has bigger screen compared to even more smaller mobile.

VoronMods, 2022-08-11:
@Pallesauka @muthijammy and others, switching back and fort between pumps will do nothing, this is an issue in the Java Code, it is a bug. @MilosKozak any info on this issue/bug?

muthijammy, 2022-08-11:
For everyone interested, I found following workaround suggested in fb works for me as well when error msg popsup:

choose a virtual pump, subtype is AAPS generic
Snooze popup error msg
Kill AAPS
Reopen AAPS. No error msg
Switch to needed pump. Error msg no longer popsup after restart of AAPS or mobile

VoronMods, 2022-08-11:
> For everyone interested, I found following workaround suggested in fb works for me as well when error msg popsup:
>
> choose a virtual pump, subtype is AAPS generic Snooze popup error msg Kill AAPS [...]

its not about suppressing the error message, but about the error itself.
Your solution doesn't work, it's not a permanent solution, the error comes on the AAPS screen and comes from AAPS

MilosKozak, 2022-10-04:
resolved by https://github.com/nightscout/AndroidAPS/commit/3f3cdde53bcff139515cf9cdfb9bb5dbd68a4ca2

closed 2022-10-04 by MilosKozak