nightscout/AndroidAPS #1960: 3.1.0.2: Important info truncated in main screen

Arne1346, 2022-07-24:
Hi,

with 3.1. some UI improvements have been applied. Unfortunately now a 6 inch phone (Xioami Mi A2) with a resolution 1080 x 2160 pixels is no longer able to display important information on the main screen due to truncation:

[image]

The red 1 in the middle of the line is the first digit of the remaining insulin units. At the moment the screenshot was taken it was 19 IU (which makes quite a different to 1 IU as displayed).

Can you rethink and enhance the new layout to make sure that all information can be displayed as before in 3.0?

thank you,
Arne

Alamo04, 2022-07-24:
Confirmed
It's because the german translation is to long (Std) instead the english (h).

Solution for development: Don't translate this information and let it stay in english.

Solution for user: If needed, set AAPS language to english.

cr582, 2022-07-25:
This issue is also seen in treatment tab among other locations. See also discussion and screenshots in #1394 (which was unfortunately and incomprehensibly closed).

jgordon5, 2022-08-03:
I have the same problem with my Honor 50 5g. Sometimes it shows full information on home screen, at other times it's truncated. I put in the screen/pixel dimensions in Android Studio but it's still often truncated and doesn't show COB in full or IOB.

MilosKozak, 2022-08-03:
> I have the same problem with my Honor 50 5g. Sometimes it shows full information on home screen, at other times it's truncated. I put in the screen/pixel dimensions in Android Studio but it's still often truncated and doesn't show COB in full or IOB.

screenshot?

jgordon5, 2022-08-03:
I'll send it to you as soon as it does it again.
Jill Gordon

jgordon5, 2022-08-05:
Hello
Here you are, the IOB is truncated and the sensitivity on the right hand
side.
Another screenshot shows insulin actually on board.

Best wishes
Jill

jgordon5, 2022-08-06:
[image]
[image]
Here are the screenshots - apologies for the delay. The first one shows what the IOB actually was. Quite often the amount of insulin in the reservoir only shows 1 figure when it should be showing 3.

jgordon5, 2022-08-06:
Here are the screenshots - apologies for the delay. The first one shows what the IOB actually was. Quite often the amount of insulin in the reservoir only shows 1 figure when it should be showing 3.  Sorry tried to make them smaller but they ended up not being. Apologies.

[image]
[image]
.

MilosKozak, 2022-08-09:
this is not resolvable. you are using too large font on small display

MilosKozak, 2022-08-09:
I updated all translation to one letter

cr582, 2022-08-09:
and here's the same device in landscape mode
[image]

MilosKozak, 2023-05-14:
improved by https://github.com/nightscout/AndroidAPS/commit/c86edb94fd59ca1105a03296a565bf405d5d2636

closed 2023-05-14 by MilosKozak