nightscout/AndroidAPS #1962: Connection to galaxy watch 4 is not recognized by AAPS 3.1.0.2

avicorp, 2022-07-24:
After update to 3.1.0.2 and re install of the wear app on Galaxy watch 4. AAPS not recognaize the connection to galaxy watch 4, the connection app is samsung wear and not google wear app.

In version 3.0.0.2 the app sends data to the watch but action from the watch didn't affect AAPS phone app.

The app was intalled on the watch using debuging via wifi.

[image]
[image]
[image]

vanelsberg, 2022-07-25:
1. I assume, in general your Watch 4 is working as expected?
2. BT and WiFi are enabled on the watch?
3. Wear app version is of the same build as AAPS?
4. How did you install the AAPS wear app on your watch?

vanelsberg, 2022-07-25:
Btw: just in case - Charge your phone, which should switch off phones battery saving mode and retry? ;-)

avicorp, 2022-07-25:
Hi, 1. Yes 2. Yes 3. Yes 4. Run from the idea using abd debug via wifi.

I am after night of charge.

andrewfaid, 2022-07-26:
Same issue on TicWatch - answers to the question are yes,yes,yes and run from AS console to physically connected (USB) watch

vanelsberg, 2022-07-26:
Did you try uninstalling - then reinstalling the wear app?

(Running watch 4 myself on 3.1.02-dev-a. No problems)

andrewfaid, 2022-07-26:
have reset watch and reinstalled twice

avicorp, 2022-07-26:
> Did you try uninstalling - then reinstalling the wear app?
> 
> (Running watch 4 myself on 3.1.02-dev-a. No problems)

How do you connect the watch to the phone?
You use samsung wearable or wear os app?

vanelsberg, 2022-07-26:
> How do you connect the watch to the phone? 
Using Samsung phone (Samsung Wear app) & Samsung Watch 4

avicorp, 2022-07-26:
> > How do you connect the watch to the phone?
> 
> Using Samsung phone (Samsung Wear app) & Samsung Watch 4

3.1.02-dev-a version uses the old version for wear connectivity, i'm still trying to pin point the problem in the new version. I will be happy for more eyes on the issue.

avicorp, 2022-07-26:
The watch app need to be sign in the same key like the app.
In case of installation using idea, debug process, you will need to set signingConfigs in "build.gradle" file of the wear.

closed 2022-07-26 by avicorp

vanelsberg, 2022-07-27:
> The watch app need to be sign in the same key like the app.

Yes, preferably build both AAPS app and wear app from same commit.
Also can recommend installing the signed .APK using Wear Installer (from Google Store). It connects to the watch over WiFi (with adb debugging over WiFi enabled), copies the .apk and installs it. Works like a charm!

gphilpott, 2022-09-03:
I have the same issue as the OP (using v3.1.0.3) with the Samsung Watch 4.

In my case the wear app uses the build variant of 'pumpcontrolRelease', using the same commit as the AAPS aps.
I transferred/installed to the watch using the freepoc.wearinstaller.

@avicorp can you confirm how you solved the issue? was it to rebuild using the commit referencing 3.1.02-dev-a?

avicorp, 2022-09-03:
Did you build a signd apk?

gphilpott, 2022-09-03:
> 

Thanks for the quick reply - yes, using the following steps:
(On the same commit as the build of the full app)
1. Run Android Studio / Build / Generate Signed Bundle or APK
2. (Leave settings as used for full app build i.e. same key store / pass)
3. Change Module to AndroidAPS.wear
4. Select Next, build variant 'pumpControlRelease'
5. Build, then transfer/install APK on to watch using the [org.freepoc.wearinstaller](https://play.google.com/store/apps/details?id=org.freepoc.wearinstaller) app

gphilpott, 2022-09-03:
(FYI - based on your comment, I am re-tracing/double checking the installs and will report back later)

avicorp, 2022-09-04:
Check it again, if that fail check the log file in the phone. 

Log file can be sent by mail in the maintenance section, you can fined it under the hamburger menu on the top left of the main screen.

gphilpott, 2022-09-04:
Thanks, I think the issue is as you suspected - a signing issue.
Running jarsigner -verify on both apk's shows that whilst the watch apk is signed, the phone (full app) was not for some reason, which I am yet to solve.

vanelsberg, 2022-09-05:
> 4. Select Next, build variant 'pumpControlRelease'

Why "pumpControlRelease"??
Just build for **fullRelease** (same goes for AAPS)

gphilpott, 2022-09-05:
> > 4. Select Next, build variant 'pumpControlRelease'
> 
> Why "pumpControlRelease"?? Just build and sign for **fullRelease** (for both AAPS and wear app)

I was looking to start open looping, but allow remote control of the pump as per instructions here: https://androidaps.readthedocs.io/en/latest/Configuration/Watchfaces.html#building-wear-os-version-of-aaps

I have been able to get the watch connected thanks to both yours and @avicorp's suggestions - **thank you both!**.
I had to manually sign the main phone app using [apksigner](https://developer.android.com/studio/command-line/apksigner) and build the watch app using 'fullRelease' to get it to connect.