nightscout/AndroidAPS #1963: Omnipod Eros / Profile Set catch 22

thecubic, 2022-07-24:
I upgraded to AndroidAPS 3.1.0 (ca3637f0f) just now during an Omnipod Eros pump switch.  When trying to activate the profile stored in the preferences, I got:
"Failed to update basal profile: the Pod is in a wrong state"

I figured that meant I needed to activate the new pump.  While at the cannula stage, the Omnipod Eros activation UI would not continue, because the basal rates were unset.

I'm not sure which action should happen first between the pod activation and profile set, but it looks as if the profile set requires an Omnipod basal push update despite a null pod state and the pod cannula insertion requires a profile to be set.

To attempt a workaround, I set the pump driver to the virtual pump, which allowed the profile switch to activate, then switched pump drivers back to Omnipod Eros.  Happily, it didn't require a new pump and it was able to resume from before I switched

AndroidAPS 3.1.0
Build: ca3637f0f-2022.07.23-16:12
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: 14.2.4

the exception error - being unable to switch profiles, occurred at 13:29:21.033 (7/24/2022).  I'm attaching some logs; I was able to find the profile switch failure event but not anything representing the UI not allowing pump progress
[export.log](https://github.com/nightscout/AndroidAPS/files/9176764/export.log)

Chikibai, 2022-07-25:
Have you try latest master 3.1.0.2 and can verify that error still exits?

vanelsberg, 2022-07-25:
Thanks for reporting:
Just to see if I understand the problem you are describing:
1. You updated to version 3.1.0 with _no active_ EROS pump.
2. On upgrading you were getting the "Failed to update basal profile: the Pod is in a wrong state" error.
Correct?

**Question**: from what version were you upgrading?

(Btw: good work around!)

vanelsberg, 2022-07-25:
Problem may be the result of fix https://github.com/nightscout/AndroidAPS/pull/1896 on handling Pod state in earlier AAPS 3.0 dev versions and should not be a problem for current release. Possible validation steps:
1. Deactivate EROS pump
2. Using AAPS 3.01 or higher, upgrade to a new version

Result:
- You should not get the "Failed to update basal profile: the Pod is in a wrong state" error.
- Pod activation should be successful

In case you do get the error use the work around as described by @thecubic:
Set the pump driver to virtual pump to allow the profile switch to activate, then switch pump driver back to Omnipod Eros.

thecubic, 2022-07-30:
I was upgrading from 3.0.0.1 I believe.  Yes, that's a correct description of the problem I'm having

closed 2022-08-06 by MilosKozak