nightscout/AndroidAPS #1965: User option to view their NS site as an alternative for the build in History Browser?

vanelsberg, 2022-07-25:
**AndroidAPS:** 3.1.0.2-dev-a
**Build:** n/a
**Flavor:** Fullrelease, developer mode

**Problem**
Current implementation for AAPS's History Browser is unbearable slow, imho unusable. Even on my Samsung S22 Ultra, one of the fastest phones in the market with fast CPU and 12Gb internal memory rendering takes minutes and/or even fails.

**Suggestion**
I think most of us are using Nightscout for history reporting.
How about giving the user the option to open their NS site page as an alternative for current History Browser?

MilosKozak, 2022-08-06:
doing this doesn't need AAPS support or?

vanelsberg, 2022-08-06:
> doing this doesn't need AAPS support or?

No, not really. But AAPS/NSclient has the NS location and secret.
So AAPS could showing NS as an in-app web page? Or maybe just a button like the one for BYODA/CGM.

Just an idea, might be overkill. User can just add a NS shortcut to the phones home screen.
And as I do not see many users voting "we need this!", feel free to close this feature request.

closed 2022-08-14 by MilosKozak